openwrt/packages #15725: boost 1.76 does not compile on older ARM

neheb, 2021-05-31:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: arm

https://downloads.openwrt.org/snapshots/faillogs/arm_fa526/packages/boost/compile.txt
https://downloads.openwrt.org/snapshots/faillogs/arm_xscale/packages/boost/compile.txt
https://downloads.openwrt.org/snapshots/faillogs/arm_arm926ej-s/packages/boost/compile.txt

ClaymorePT, 2021-06-03:
@neheb It seems that regex library is getting built, something that is no longer necessary. Regex is now a header-only library and building it should be disabled, since it was deprecated for C++11 and forward. The regex library will only be supported for C++03, which is no longer OpenWRT case, right?

neheb, 2021-06-03:
I mean... OpenWrt supports C+(11 yes.

ClaymorePT, 2021-06-03:
Well, it seems that the library became a header-only, but for source `boost/regex/v5/perl_matcher_non_recursive.hpp`, someone forgot to update the mutex type along with it's initialization. 
I've opened an issue upstream https://github.com/boostorg/regex/issues/131 and  I'll devise a patch to fix this for OpenWRT.

closed 2021-06-07 by neheb