openwrt/packages #15765: stubby: Disable sending DNS requests to ISP provided DNS servers guidance

jamesmacwhite, 2021-06-03:
Maintainer: @jonathanunderwood 

Description:

The guidance on disabling ISP DNS servers is potentially bad in the README, setting dnsmasq to not use the resolv.conf is fine, but modifying the DNS servers on the WAN interface itself has found to cause grief based on community feedback: https://forum.openwrt.org/t/tutorial-no-cli-configuring-dns-over-tls-with-luci-using-stubby-and-dnsmasq/29143/34

If there are concerns about DNS leaks, it is probably better to utilise the firewall to prevent traffic to DNS servers, or intercepting DNS requests.

jonathanunderwood, 2021-06-13:
The README is correct. The correct way to address this is to set a fallback for if TLS is not available yet:

```
config stubby 'global'
       option manual '0'
       option trigger 'wan'
       # option triggerdelay '2'
       list dns_transport 'GETDNS_TRANSPORT_TLS'
       list dns_transport 'GETDNS_TRANSPORT_UDP'
       list dns_transport 'GETDNS_TRANSPORT_TCP'
...
```

This way, if NTP is not setup (such that TLS would fail), stubby will fall back to a non-TLS connection to the specified DNS servers.

Your proposal would also fix the problem, with the result of DNS leaks to a DNS provider not requested by the user.

jonathanunderwood, 2021-06-13:
See also:
- https://github.com/getdnsapi/stubby/issues/274
- https://github.com/getdnsapi/stubby/issues/146 (note mention of opportunistic configuration, which is what I describe above)
- https://dnsprivacy.org/wiki/display/DP/Configuring+Stubby#ConfiguringStubby-OpportunisticDoTtoyourlocalresolver

jonathanunderwood, 2021-06-13:
Also, to note, I am no longer maintaining the stubby package in OpenWRT.

jamesmacwhite, 2021-10-05:
@jonathanunderwood Hi, sorry I never replied to this! Thank you for comprehensive response.

It would appear opportunistic mode is likely a cleaner configuration fix.

I revisited this I encountered the exact scenario of NTP not being able to sync time, because DNS resolution of the NTP pool addresses through stubby failed, because of the system time being wrong and hence breaking TLS. (inception!). I get the impression it's not always a problem if NTP manages to update using plain DNS before stubby is used.

I guess the key thing here is opportunistic mode allows the best of both worlds with reduced privacy, which may or not be desired. One area might be how would you know when the stubby TLS connection is broken and it's falling back, would this be logged anywhere to capture, perhaps to send an alert or email.

A dirty hack might be to do something like:

```
list server '/0.openwrt.pool.ntp.org/1.openwrt.pool.ntp.org/2.openwrt.pool.ntp.org/3.openwrt.pool.ntp.org/1.1.1.1'
```

So you are only sending those NTP DNS names through plain DNS without configuring opportunistic mode. Although, that's just for NTP, I'm sure there are other edge cases of DNS names that might need to be allowed through plain DNS until stubby is ready on a full start up.

I'll close the issue itself as you have clearly outlined a solution to implement in the configuration which is a supported use case.

closed 2021-10-05 by jamesmacwhite