openwrt/packages #15768: freeradius3: update to version 3.0.22

stertingen, 2021-06-03:
Maintainer(s): @odmdas @neheb 
Environment: OpenWRT 19.07 on Archer C7

Description: Freeradius has been updated to version 3.0.22. This version is particularly interesting for me as it fixes issues with the FreeIPA LDAP server.

See also:
https://github.com/FreeRADIUS/freeradius-server/releases/tag/release_3_0_22
https://github.com/FreeRADIUS/freeradius-server/issues/2649

CodeFetch, 2021-06-17:
@stertingen https://github.com/openwrt/packages/pull/15896
Please test it. I don't use radius atm. To build a firmware with my version of the package just make sure your OpenWrt build directory is clean, uncomment the first line of your feed.conf.default and change it to:
`src-git packages https://github.com/codefetch/packages-3.git;patch-12`
and then do a `scripts/feeds update -a && scripts/feeds install -a`.

hnyman, 2023-01-14:
Closing this request, as there is the open (but apparently stale) PR #15896 about the requested version bump.

closed 2023-01-14 by hnyman