openwrt/packages #15781: luci-app-acme: Add config options for "--deploy-hook"

KairuByte, 2021-06-05:
Maintainer: @tohojo
Environment: 
- ARMv7 Processor rev 1 (v7l)
- Linksys WRT3200ACM
- OpenWrt 21.02.0-rc2 r16122-c2139eef27 / LuCI openwrt-21.02 branch git-21.151.34805-8e14a08

Description:
Unless I am mistaken, there is an acme command line option which cannot currently be utilized by luci-app-acme, which would allow for more flexible deployment of certs. Namely [--deploy-hook](https://github.com/acmesh-official/acme.sh/wiki/deployhooks). This would likely be useful to many, as it has built in solutions for a large number of deployment situations.

tohojo, 2021-06-05:
KairuByte ***@***.***> writes:

> Maintainer: @tohojo
> Environment:
> - ARMv7 Processor rev 1 (v7l)
> - Linksys WRT3200ACM
> - OpenWrt 21.02.0-rc2 r16122-c2139eef27 / LuCI openwrt-21.02 branch git-21.151.34805-8e14a08
>
> Description:
> Unless [...]

"would likely be useful" - do you have a use case in mind?

KairuByte, 2021-06-08:
A few, such as the placement and/or creation of specific very files in certain places. One such would be haproxy, though that can be accomplished with some linking, though it's not the cleanest.

The main use case I had in mind was deployment over SSH to a secondary source, to prevent an excess of calls to LE servers.

tohojo, 2021-06-11:
KairuByte ***@***.***> writes:

> A few, such as the placement and/or creation of specific very files in
> certain places. One such would be haproxy, though that can be
> accomplished with some linking, though it's not the cleanest. [...]

Hmm, right, well I'm not opposed to exposing the hook as a luci config
option; care to open a PR? :)

KairuByte, 2021-06-11:
If I can figure it out, I will. The parts I need to work it out should already be in place I think, just needs some copy/paste and sanity checking.