openwrt/packages #15783: NGINX stops for "504 Gateway Time-out" if HTTPS-DNS-PROXY servise in on and WAN is off

desigabri, 2021-06-05:
NGINX Maintainer: Thomas Heil heil@terminal-consulting.de, Ansuel Smith ansuelsmth@gmail.com
NGINX-UTIL Maintainer: Peter Stadler peter.stadler@student.uibk.ac.at

Environment: Qualcomm Atheros IPQ806X, Archer C2600 v1, "21.02, Master

Description:

THIS TROUBLE HAPPENS FOR nginx IF HTTP-DNS-Proxy PACKAGE IS INSTALLED AND ABILITATED

If the router isn't connected to internet (or to WAN?) nginx try starting but then FAILS showing **"504 Gateway Time-out"**

The troble can be solved just connecting the WAN cable (connected to internet) and  then disabling HTTP-DNS-PROXY service. Doing this the router become accessible also if WAN is disconnected.

This is way I'm going to open a ticket to HTTP-DNS-PROXY  asking to disable the service for defautl settings.
I'll open  a ticket also to NGINX  package because may be this is a trouble of NGINX that isn't able to bypass the lack of WAN.

BKPepe, 2021-06-08:
Related to #15784, closing. There is no need to have two opened issues for one issue.

closed 2021-06-08 by BKPepe

desigabri, 2021-06-08:
I opened two tickets because I don't know if the problem come from NGINX or from HTTP-DNS-PROXY and the rules ask to refer the ticket to a defined package

desigabri, 2021-09-26:
Finally I had two problems at the same time that comfused me: HTTPS-DNS-PROXY solved the problems by itself (#16252) and About NGINX I found the FIX in #14465