openwrt/packages #15784: HTTP-DNS-PROXY SERVICE STOPS nginx SERVER IF HTTP-DNS-PROXY IS ENABLED AND WAN IS OFF (WITHOUT CONNECTION)

desigabri, 2021-06-05:
http-dns-proxy Maintainer:  Stan Grishin stangri@melmac.net

Environment: Qualcomm Atheros IPQ806X, Archer C2600 v1, "21.02, Master

Description:

THIS TROUBLE HAPPENS FOR nginx IF HTTP-DNS-Proxy PACKAGE IS INSTALLED AND ABILITATED

If the router isn't connected to internet (or to WAN?) nginx try starting but then FAILS showing **"504 Gateway Time-out"**

The troble can be solved just connecting the WAN cable (connected to internet) and  then disabling HTTP-DNS-PROXY service. Doing this the router become accessible also if WAN is disconnected.

This is way I'm going to open a ticket to HTTP-DNS-PROXY  asking to disable the service for defautl settings.
I'll open  a ticket also to NGINX  package because may be this is a trouble of NGINX that isn't able to bypass the lack of WAN.

I THINK IT COULD BE USEFULL AT THE FIRST TIME, MAKING HTTP-DNS-PROXY SERVICE DISABLED FOR DEFAULT

desigabri, 2021-09-26:
Finally I had two problems at the same time that comfused me: HTTPS-DNS-PROXY solved the problems by itself (#16252) and About NGINX I found the FIX in #14465

stangri, 2021-10-01:
Closing as reported fixed.

closed 2021-10-01 by stangri