openwrt/packages #15791: nginx-util: 21.02-rc2, fails to build

nickberry17, 2021-06-07:
Please make sure that the issue subject starts with `<package-name>: `

Maintainer: @peter-stadler 
Environment: ipq4019, Telco X1 Pro, 21.02-rc2

Description:
Package fails to build.

```
/home/code/openwrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-8.4.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-g++ -DNO_PCRE -DVERSION=1.5  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5=nginx-util-1.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG   -Os -Wall -Werror -Wextra -g3 -Wno-unused-parameter -Wmissing-declarations -Wshadow -std=gnu++17 -o CMakeFiles/nginx-ssl-util-nopcre.dir/nginx-util.cpp.o -c /home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-util.cpp
In file included from /home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-util.cpp:4:
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp: In lambda function:
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:169:40: error: attributes at the beginning of statement are ignored [-Werror=attributes]
                 case '^': ret += '\\'; [[fallthrough]];
                                        ^
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:170:27: error: attributes at the beginning of statement are ignored [-Werror=attributes]
                 case '_': [[fallthrough]];
                           ^
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp: In lambda function:
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
                 case '^': ret += '\\'; [[fallthrough]];
                           ~~~~^~~~~~~
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:170:17: note: here
                 case '_': [[fallthrough]];
                 ^~~~
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp: In lambda function:
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
                 case '^': ret += '\\'; [[fallthrough]];
                           ~~~~^~~~~~~
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:170:17: note: here
                 case '_': [[fallthrough]];
                 ^~~~
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp: In lambda function:
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
                 case '^': ret += '\\'; [[fallthrough]];
                           ~~~~^~~~~~~
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:170:17: note: here
                 case '_': [[fallthrough]];
                 ^~~~
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp: In lambda function:
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
                 case '^': ret += '\\'; [[fallthrough]];
                           ~~~~^~~~~~~
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:170:17: note: here
                 case '_': [[fallthrough]];
                 ^~~~
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp: In lambda function:
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
                 case '^': ret += '\\'; [[fallthrough]];
                           ~~~~^~~~~~~
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:170:17: note: here
                 case '_': [[fallthrough]];
                 ^~~~
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp: In lambda function:
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
                 case '^': ret += '\\'; [[fallthrough]];
                           ~~~~^~~~~~~
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:170:17: note: here
                 case '_': [[fallthrough]];
                 ^~~~
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp: In lambda function:
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
                 case '^': ret += '\\'; [[fallthrough]];
                           ~~~~^~~~~~~
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:170:17: note: here
                 case '_': [[fallthrough]];
                 ^~~~
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp: In lambda function:
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
                 case '^': ret += '\\'; [[fallthrough]];
[... 37 lines cut ...]
                 case '_': [[fallthrough]];
                 ^~~~
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp: In lambda function:
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
                 case '^': ret += '\\'; [[fallthrough]];
                           ~~~~^~~~~~~
/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/nginx-ssl-util.hpp:170:17: note: here
                 case '_': [[fallthrough]];
                 ^~~~
cc1plus: all warnings being treated as errors
make[5]: *** [CMakeFiles/nginx-ssl-util-nopcre.dir/build.make:82: CMakeFiles/nginx-ssl-util-nopcre.dir/nginx-util.cpp.o] Error 1
make[5]: Leaving directory '/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5'
make[4]: *** [CMakeFiles/Makefile2:97: CMakeFiles/nginx-ssl-util-nopcre.dir/all] Error 2
make[4]: Leaving directory '/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5'
make[3]: *** [Makefile:149: all] Error 2
make[3]: Leaving directory '/home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5'
make[2]: *** [Makefile:121: /home/code/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nginx-util-1.5/.built] Error 2
make[2]: Leaving directory '/home/code/openwrt/feeds/packages/net/nginx-util'
time: package/feeds/packages/nginx-util/compile#3.79#0.14#3.76
    ERROR: package/feeds/packages/nginx-util failed to build.
make[1]: *** [package/Makefile:114: package/feeds/packages/nginx-util/compile] Error 1
make[1]: Leaving directory '/home/code/openwrt'
make: *** [/home/code/openwrt/include/toplevel.mk:230: package/nginx-util/compile] Error 2

```

peter-stadler, 2021-06-07:
Does the following helps?
https://forum.openwrt.org/t/error-compiling-nginx-for-v21-02-0-rc2/98082/6

neheb, 2021-06-07:
Hrm thought I backported it.

peter-stadler, 2021-06-07:
I did see your back port, but in the forum @hnyman  mentioned that later fixes are not propagated to the rc2 ...
I did some tests before using `break;`
Putting `[[fallthrough]];` on a new line did not work. Using `fallthrough;` instead worked, but the standard wants it with brackets (and that worked before, I do not know when it stopped working) ...
But all this will not help if the code for the rc2 cannot be changed.

hnyman, 2021-06-08:
> mentioned that later fixes are not propagated to the rc2 ...

Normal packages are constantly rebuilt, so fixes get propagated in the 21.02 branch. The trouble I have been debugging only concerns target-specific packages and/or depdendencies from those to ABI versioned normal packages.

Note that if you build from the 21.02.0-rc2 tag instead of the 21.02 branch HEAD, you will not see the fixes, as you have feeds locked to the rc2 moment.   Check your feeds.conf.default

hnyman, 2023-01-14:
Seems to build ok for 21.02: available as download for 21.02.5

closed 2023-01-14 by hnyman