openwrt/packages #15805: netdata: compiles with disable cloud argument

lucki1000, 2021-06-08:
Netdata: Is there a reason why netdata is compiled with `--disable-cloud`?
source: [https://github.com/openwrt/packages/blob/master/admin/netdata/Makefile](https://github.com/openwrt/packages/blob/master/admin/netdata/Makefile)

Maintainer: @BKPepe 
My Environment:  `Archer C7 V5; Linux OpenWrt 5.4.119; Arch=mips; OpenWrt 21.02.0-rc2`

Regards Lucki1000

BKPepe, 2021-06-08:
Hello @lucki1000,

Yes, take a look here: https://github.com/openwrt/packages/pull/12164#issuecomment-638004279
You can of course propose a patch. :+1:

underhood, 2021-06-10:
@BKPepe by default no Cloud connection will be made until you claim the agent (user has to do that manually). Any other behavior would be taken as a very very serious bug by us.

Of course, the final decision is up to you/OpenWRT team. In case there are any technical issues with enabling cloud feel free to ping me and I will try to assist.

As per one of your comments on #12164 some distro packages disable Cloud due to us bundling libmosquitto which we used with a patch (distros have policy against bundling libraries which are already present as separate package in the system). This is one of main reasons also for us to implement new stack to do the cloud connection called ACLK-NG which you can use in case this is case also for OpenWRT.

diizzyy, 2021-06-11:
Again, size is the main concern but if someone wants to step in and provide "flavours" of a package feel free to do so.

bolemo, 2022-01-07:
Hello all,

We are now in 2022.
The netdata cloud node uses a new claiming link: https://learn.netdata.cloud/docs/agent/aclk

I use netdata from entware-ng (mostly based on the openwrt one).
The latest version available from opkg (1.30.1 at this time) does create the /opt/var/lib/netdata/cloud.d/ folder (tat is empty), and the netdata --help does list the option -W "claim -token=TOKEN -rooms=ROOM1,ROOM2"

However, the cloud feature is still intentionally disabled:

```
root@HERMES:/tmp/mnt/sda1/entware/var/lib/netdata/cloud.d$ netdata -W buildinfo
Version: netdata v1.30.1
Configure options:  '--target=arm-openwrt-linux' '--host=arm-openwrt-linux' '--build=x86_64-pc-linux-gnu' '--program-prefix=' '--program-suffix=' '--prefix=/opt' '--exec-prefix=/opt' '--bindir=/opt/bin' '--sbindir=/opt/sbin' '--libexecdir=/opt/lib' '--sysconfdir=/opt/etc' '--datadir=/opt/share' '--localstatedir=/opt/var' '--mandir=/opt/man' '--infodir=/opt/info' '--disable-nls' '--with-zlib' '--with-math' '--disable-x86-sse' '--enable-lto' '--disable-ebpf' '--without-libcap' '--disable-https' '--disable-dbengine' '--disable-plugin-nfacct' '--disable-plugin-freeipmi' '--disable-plugin-cups' '--disable-plugin-xenstat' '--disable-backend-prometheus-remote-write' '--disable-unit-tests' '--disable-cloud' 'build_alias=x86_64-pc-linux-gnu' 'host_alias=arm-openwrt-linux' 'target_alias=arm-openwrt-linux' 'CC=arm-openwrt-linux-gnueabi-gcc' 'CFLAGS=-pipe -mcpu=cortex-a15 -mfpu=neon-vfpv4 -funsafe-math-optimizations -mtune=cortex-a15 -fcommon -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffunction-sections -fdata-sections -O3 ' 'LDFLAGS=-Wl,--dynamic-linker=/opt/lib/ld-linux.so.3 -Wl,-rpath=/opt/lib -L/home/voxel/Entware/staging_dir/toolchain-arm_cortex-a15_gcc-9.4.0_glibc-2.27_eabi/lib -Wl,--gc-sections ' 'CPPFLAGS=-I/home/voxel/Entware/staging_dir/toolchain-arm_cortex-a15_gcc-9.4.0_glibc-2.27_eabi/include ' 'CXX=arm-openwrt-linux-gnueabi-g++' 'CXXFLAGS=-pipe -mcpu=cortex-a15 -mfpu=neon-vfpv4 -funsafe-math-optimizations -mtune=cortex-a15 -fcommon -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffunction-sections -fdata-sections -O3 ' 'PKG_CONFIG=/home/voxel/Entware/staging_dir/host/bin/pkg-config' 'PKG_CONFIG_PATH=/home/voxel/Entware/staging_dir/target-arm_cortex-a15_glibc-2.27_eabi/opt/lib/pkgconfig:/home/voxel/Entware/staging_dir/target-arm_cortex-a15_glibc-2.27_eabi/opt/share/pkgconfig' 'PKG_CONFIG_LIBDIR=/home/voxel/Entware/staging_dir/target-arm_cortex-a15_glibc-2.27_eabi/opt/lib/pkgconfig:/home/voxel/Entware/staging_dir/target-arm_cortex-a15_glibc-2.27_eabi/opt/share/pkgconfig'
Features:
    dbengine:                NO
    Native HTTPS:            NO
    Netdata Cloud:           NO (by user request)
    TLS Host Verification:   YES
Libraries:
    jemalloc:                NO
    JSON-C:                  YES
    libcap:                  NO
    libcrypto:               YES
    libm:                    YES
    LWS:                     YES shared-lib
    mosquitto:               YES
    tcalloc:                 NO
    zlib:                    YES
Plugins:
    apps:                    YES
    cgroup Network Tracking: YES
    CUPS:                    NO
    EBPF:                    NO
    IPMI:                    NO
    NFACCT:                  NO
    perf:                    YES
    slabinfo:                YES
    Xen:                     NO
    Xen VBD Error Tracking:  NO
Exporters:
    AWS Kinesis:             NO
    GCP PubSub:              NO
    MongoDB:                 NO
    Prometheus Remote Write: NO
```
Any chance to have this working now that the ACKL is simpler ans requires less dependencies?
By default, it could be turn off in the netdata.conf, but enabling the cloud would allow users to turn it on if they wish.

Thank you

BKPepe, 2022-01-07:
> We are now in 2022.

Well, yes. That's true and because of that, we need to take care of our privacy more than in the past.

stintel, 2022-01-07:
As suggested before, if this is a problem for you, submit a PR that adds a new flavour of the package with cloud support enabled.

closed 2022-01-07 by stintel