openwrt/packages #15807: bluld-1.1.1 fails to build

ByteEnable, 2021-06-08:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:
make[6]: Entering directory '/home/ByteEnable/temp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/bluld-1.1.1'
[ 11%] Building CXX object CMakeFiles/bluld.dir/blinkstick-userspace-led-daemon/src/LEDBinding.cpp.o
/home/ByteEnable/temp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/bluld-1.1.1/blinkstick-userspace-led-daemon/src/LEDBinding.cpp: In member function 'void BlinkstickUserspace::LEDBinding::registerUserSpaceLED()':
/home/ByteEnable/temp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/bluld-1.1.1/blinkstick-userspace-led-daemon/src/LEDBinding.cpp:57:10: error: 'char* strncpy(char*, const char*, size_t)' specified bound 64 equals destination size [-Werror=stringop-truncation]
   strncpy(dev.name, mName.c_str(), LED_MAX_NAME_SIZE);
   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[6]: *** [CMakeFiles/bluld.dir/build.make:95: CMakeFiles/bluld.dir/blinkstick-userspace-led-daemon/src/LEDBinding.cpp.o] Error 1
make[6]: Leaving directory '/home/ByteEnable/temp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/bluld-1.1.1'
make[5]: *** [CMakeFiles/Makefile2:95: CMakeFiles/bluld.dir/all] Error 2
make[5]: Leaving directory '/home/ByteEnable/temp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/bluld-1.1.1'
make[4]: *** [Makefile:149: all] Error 2
make[4]: Leaving directory '/home/ByteEnable/temp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/bluld-1.1.1'
make[3]: *** [Makefile:50: /home/ByteEnable/temp/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/bluld-1.1.1/.built] Error 2
make[3]: Leaving directory '/home/ByteEnable/temp/openwrt/feeds/packages/utils/bluld'
time: package/feeds/packages/bluld/compile#0.41#0.07#0.47
    ERROR: package/feeds/packages/bluld failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/bluld/compile] Error 1
make[2]: Leaving directory '/home/ByteEnable/temp/openwrt'
make[1]: *** [package/Makefile:110: /home/ByteEnable/temp/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ByteEnable/temp/openwrt'
make: *** [/home/ByteEnable/temp/openwrt/include/toplevel.mk:230: world] Error 2

```
Formating code blocks by wrapping them with pairs of ```
```

neheb, 2021-06-08:
report it here: https://github.com/ktgeek/bluld/issues

neheb, 2021-06-08:
quick way to fix is to replace

```
LED_MAX_NAME_SIZE
```

with

```
LED_MAX_NAME_SIZE -1
```

neheb, 2021-06-08:
oh @ktgeek is active here.

ktgeek, 2021-06-10:
oh man, facepalm to that coding error.  I wonder why that hasn't pinged up for me before.  Did you have any settings that might bright that out more?

I'll get that fixed up.

PR #15820 (2021-06-10): bluld: fix compilation issues for nls and string truncation
Fixes #15807
Fixes #15818

Maintainer: me
Compile tested: aarch64_cortex-a53 master, ath79 archer c7v2 master
Run tested: ath79 archerc7v2 master

Description:
Fixes two issues compilation issues reported to the openwrt/packages GitHub repository.
commit 4ec036a8 (2021-06-10): bluld: fix compilation issues for nls and string truncation

Fixes #15807
Fixes #15818
closed 2021-06-10 by neheb