openwrt/packages #15818: bluld: undefined reference to libiconv

BKPepe, 2021-06-10:
# Issue template (remove lines from top till here)

Maintainer: @ktgeek 
Environment: Turris MOX, aarch64_cortex-a53 (Turris MOX), OpenWrt master

Description:

```
/foo/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: warning: libiconv.so.2, needed by /foo/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libhidapi-libusb.so, not found (try using -rpath or -rpath-link)
/foo/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /foo/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libhidapi-libusb.so: undefined reference to `libiconv'
/foo/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /foo/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libhidapi-libusb.so: undefined reference to `libiconv_close'
/foo/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /foo/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libhidapi-libusb.so: undefined reference to `libiconv_open'
collect2: error: ld returned 1 exit status
```

ktgeek, 2021-06-10:
I'll check that out and see if I can fix it.  But at first pass, it looks, though, as if this problem might be in the hidapi (or libusb) package not specifying a dependency it has.

ktgeek, 2021-06-10:
I can recreate the compile error in openwrt on aarch64_cortex-a53 if I include full language support...  Looks like I missed adding nls.mk to the build.  I'll have a fix up shortly.

PR #15820 (2021-06-10): bluld: fix compilation issues for nls and string truncation
Fixes #15807
Fixes #15818

Maintainer: me
Compile tested: aarch64_cortex-a53 master, ath79 archer c7v2 master
Run tested: ath79 archerc7v2 master

Description:
Fixes two issues compilation issues reported to the openwrt/packages GitHub repository.
commit 4ec036a8 (2021-06-10): bluld: fix compilation issues for nls and string truncation

Fixes #15807
Fixes #15818
closed 2021-06-10 by neheb