openwrt/packages #15843: stubby: fails to build

anomeome, 2021-06-12:
```
install -m0600 ./files/stubby.conf /home/kc/wrtpac/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/stubby-0.4.0/ipkg-arm_cortex-a9_vfpv3/stubby/etc/config/stubby
find /home/kc/wrtpac/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/stubby-0.4.0/ipkg-arm_cortex-a9_vfpv3/stubby -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package stubby is missing dependencies for the following libraries:
libunbound.so.8
make[3]: *** [Makefile:78: /home/kc/wrtpac/source/bin/targets/mvebu/cortexa9/packages/stubby_0.4.0-1_arm_cortex-a9_vfpv3.ipk] Error 1
```

anomeome, 2021-06-13:
resolved

closed 2021-06-13 by anomeome

jonathanunderwood, 2021-06-13:
@neheb this is the kind of thing that should have been tested for before merging. I see you have now fixed it (wrongly), but this is an example of why you should actually wait for the package maintainer to approve PRs.