openwrt/packages #15854: OpenVPN is not working with servers using TCP protocol

weikai, 2021-06-13:
OpenVPN client is not working with servers using TCP protocol on OpenWRT Snapshot.  It connects fine without any issues but can't transfer any data.  However, it works when I reduce the tcpmss to 59 but performance is really bad.

iptables -t mangle -A POSTROUTING -o tun3 -p tcp -m tcp --tcp-flags SYN,RST SYN -m comment --comment "!fw3: MTU fix" -j TCPMSS --set-mss 59

neheb, 2021-06-13:
ping @mkrkn

mkrkn, 2021-06-16:
I can pass traffic over a TCP tunnel just fine, without any form of size tuning. You need to provide more information:
- Your OpenVPN configuration
- Your WAN interface type (PPPoE, Ethernet, etc.) may be useful

That said, if its sizing issues I may not be able to reproduce your issue either way. You may have better results setting MTU tuning in OpenVPN itself rather than via the firewall. I believe the tun_mtu option is what you need.

weikai, 2021-06-17:
The following is my VPN client config.  It's working fine on OpenWRT 19.07 but run into MTU issue on the newest OpenWRT version.  WAN is FIOS with direct Ethernet connection.  I can't set MTU that low in OpenVPN config.  It starts complaining and refuses to connect when MTU is around or lower than 500. 

`config interface 'wan'
	option device 'eth0'
	option proto 'dhcp'`

`config openvpn 'vpn'
	option config '/etc/openvpn/vpn.ovpn'
	option enabled '1'

`

`client
comp-lzo yes
nobind
persist-key
persist-tun
auth-nocache
dev tun3
verb 3
keepalive 10 60
data-ciphers-fallback 'AES-128-CBC'

#tun-mtu 1472

remote vpn.domain 443
proto tcp
key-direction 1
remote-cert-tls server
verb 3
resolv-retry infinite
auth SHA256
reneg-sec 0
user nobody
group nogroup
auth-user-pass /etc/openvpn/vpn.auth

#route-noexec
#script-security 2
#up "/etc/openvpn/vpn/updown.sh up"
#down "/etc/openvpn/vpn/updown.sh down"

<ca>
-----BEGIN CERTIFICATE-----
cert removed
-----END CERTIFICATE-----
</ca>
<cert>
-----BEGIN CERTIFICATE-----
key removed
-----END CERTIFICATE-----
</cert>
<key>
-----BEGIN PRIVATE KEY-----
key removed
-----END PRIVATE KEY-----
</key>
<tls-auth>
#
# 2048 bit OpenVPN static key
#
-----BEGIN OpenVPN Static key V1-----
key removed
-----END OpenVPN Static key V1-----
</tls-auth>`

mkrkn, 2021-06-17:
How have you concluded that this is specific to TCP connections?

`comp-lzo yes`

Are you using image and packages from downloads.openwrt.org, or have you built your own? LZO support is disabled in the OpenWrt OpenVPN package since 2.5.0. 

If you need LZO compression on current OpenWrt you need to build your own OpenVPN package and enable the feature.
If you only need some form of compression LZ4 is still available, you will need LZ4 support and configuration (available since OpenVPN 2.4.0) on the other end.

I recommend disabling compression completely.

mmokrejs, 2021-06-25:
The compression and cipher negotiation between client server has been changed in openvpn-2.6.x and also in 2.7.x series. OpenWrt 21.02.0-rc3 provides openvpn-2.52 which is still way too old. Is there any change we could get an upgrade of the package before the new release?

pezz, 2021-06-28:
I'm trying rc3 right now and there's definitely something not right with this version.

I connect to a UDP based openvpn server at work and TCP things like ssh over it will return small amounts of data, e.g. if I telnet to a server on port 22 over the VPN I get the version string in response, but trying to establish a normal ssh session doesn't return data and just hangs unless I set a really small mssfix.

I'm currently using "mssfix 256" in my config on rc3.

On 19.07.7 I never had to use mssfix or any weird MTU settings.

pezz, 2021-06-29:
Is there any chance of getting openvpn on to a later version?

openvpn (both openssl and mbedtls) 2.5.3-1 just don't work properly.

I'm happy to keep testing openwrt 21.02 rc3 but it's getting to a point where I'm just going to go back to 19.07 if there's no way forward here.

mkrkn, 2021-06-29:
> 
> 
> Is there any chance of getting openvpn on to a later version?

No. 2.5.3 **is** the latest version, released less than two weeks ago. Of course, I will submit updates as future releases arrive. There is no OpenVPN 2.6 or 2.7 series, @mmokrejs comment makes no sense.

> there's no way forward here.

There is no way forward here indeed, as little to no useful information is posted. After suggesting a change to @weikai's configuration there has not been any response. No one else has shared their local or remote configs, or turned up ``verb`` and posted logs for that matter.

LZO compression in OpenVPN is disabled by default from OpenWrt 21.02 onwards. I suspect that this is related, and that the MSS/MTU hacks cause packets to become too small to be compressed even if the remote end expects LZO compressed data.

pezz, 2021-06-29:
Yeah, fair enough too, I didn't contribute much to this. My work's server does indeed have comp-lzo enabled so perhaps that's something I can explore, but I thought it wouldn't even connect if one side didn't like a setting.

Anyway, nothing I tried was able to resolve it so out of frustration I've gone back to 19.07 and its version of openvpn.

mmokrejs, 2021-06-30:
> > Is there any chance of getting openvpn on to a later version?
> 
> No. 2.5.3 **is** the latest version, released less than two weeks ago. Of course, I will submit updates as future releases arrive. There is no OpenVPN 2.6 or 2.7 series, @mmokrejs comment makes no sense.

Ah, sorry, I meant 2.4.6 and 2.4.7, at around these there were some changes related to negotiation. Sorry for the false message.

pezz, 2021-07-03:
Sorry to be a pain, but can I just ask why the ability to enable lzo isn't in the newer version?

Doesn't seem ideal to expect people to have to compile their own version, just for what seems to be a pretty common setting.

champtar, 2021-07-03:
@pezz the relevant commit https://github.com/openwrt/packages/commit/e4376793b4e093089543cb1bad64eef34ed25eca
I'm not using openvpn anymore but comp-lzo is often enabled (for bad reasons) and people don't always have control of the server side.
@mkrkn Maybe a middle ground could be to enable lzo with the openssl variant, because if you are using openssl you already don't care about space usage that much.

champtar, 2021-07-03:
And you could even finish removing LZ4 from mbedtls/wolfsl versions

champtar, 2021-07-13:
#16035 is merged into maste, #16113 opened for 21.02

champtar, 2021-07-13:
For all people having issues, please test openvpn-openssl_2.5.3-2 once available (don't know how fast the packages are built)