openwrt/packages #15859: nginx-all-module: does not meet dependencies for luci-ssl-nginx

rguerrab, 2021-06-14:
Tried to run LuCI off nginx with SSL on my box running:
OpenWrt 19.07.5 r11257-5090152ae3 / LuCI friendlywrt-v19.07.5 branch git-20.358.36252-85ff2c9
but can't install luci-ssl-nginx after first installing nginx-all-module because opkg wants to install nginx-ssl as a dependency to luci-ssl-nginx: which should already be provided by nginx-all-module.

CodeFetch, 2021-06-28:
That's strange, because nginx-all-module provides nginx-ssl...
@peter-stadler Can you have a look?

peter-stadler, 2021-06-30:
I will take a look at the weekend. Thanks for the notification :-)

peter-stadler, 2021-07-04:
In the 19.07 it provides only nginx not nginx-ssl, made a PR now.

rguerrab, 2021-07-05:
Awesome, thanks so much.

hnyman, 2023-01-14:
Fixed by #16036 in September 2021.

closed 2023-01-14 by hnyman