openwrt/packages #15871: python3: build failure when building custom firmware with python package

anyuta1166, 2021-06-15:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @commodo @jefferyto
Environment: x86-64, r16955-946019637e

Description:

I’m unable to build a custom firmware with python3-light package enabled.

I’m encountering the build failure:

```
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libpython3--3.9 for python3-base
 * pkg_hash_fetch_best_installation_candidate: Packages for python3-base found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package python3-base.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for python3-light:
 *      libpython3--3.9
 * opkg_install_cmd: Cannot install package python3-light.
make[2]: *** [package/Makefile:70: package/install] Error 255
```

There is some dependency issue. The libpython3 package is built as libpython3-3.9, but python3-base wants libpython3–3.9 (note the two dashes).

I'm not sure where to post this issue (here or to bugs.openwrt.org) because after some investigation I've found that the build is broken after this commit: f6a03bff5bccdbf9165087bccbb35095903d05c6 build: prepend ABI suffixes with a dash if package name ends with digit
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=f6a03bff5bccdbf9165087bccbb35095903d05c6

Reverting this commit allows to successfully build the firmware.

I don't see anything wrong in python package makefile. So I really don't know is this a base system issue or a package issue.

commodo, 2021-06-15:
@jow-    ping

is this with a clean checkout?
i will try a build now;
but, the issue is not Python directly, but more with some OpenWrt core logic

anyuta1166, 2021-06-15:
> is this with a clean checkout?

Yes.

jow-, 2021-06-15:
Will look into it.

jow-, 2021-06-15:
Fix pushed.

commodo, 2021-06-15:
> Fix pushed.

works for me

@anyuta1166 ?

anyuta1166, 2021-06-15:
Works for me too.

closed 2021-06-15 by anyuta1166