openwrt/packages #15903: irqbalance - (21.02.0-rc3) Daemon couldn't be bound to the file-based socket.

Strykar, 2021-06-18:
Maintainer: @hnyman 
Environment: (`x86_64`, `OpenWrt 21.02.0-rc3 r16172-2aba3e9784`)

Description:
`irqbalance` fails to start and logs:
```
Fri Jun 18 17:11:47 2021 daemon.warn /usr/sbin/irqbalance: Daemon couldn't be bound to the file-based socket.
```
This issue was reported in June 2020 and the `/run/irqbalance/` path issue was patched out, is this a regression?
See https://github.com/openwrt/packages/issues/12377 and https://github.com/openwrt/packages/commit/4f0c847828b9602d12d13059a5e8bb159120dcb7.

Running strace shows it failing because `/etc/ld-musl-x86_64.path` does not exist:
```
# strace /usr/sbin/irqbalance 
execve("/usr/sbin/irqbalance", ["/usr/sbin/irqbalance"], 0x7ffeb6885d70 /* 13 vars */) = 0
arch_prctl(ARCH_SET_FS, 0x7fe825254d48) = 0
set_tid_address(0x7fe825255334)         = 20526
open("/etc/ld-musl-x86_64.path", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libgcc_s.so.1", O_RDONLY|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0644, st_size=73744, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\r0\0\0\0\0\0\0"..., 960) = 960
mmap(NULL, 81920, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7fe8251cd000
mmap(0x7fe8251d0000, 49152, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0x3000) = 0x7fe8251d0000
mmap(0x7fe8251dc000, 12288, PROT_READ, MAP_PRIVATE|MAP_FIXED, 3, 0xf000) = 0x7fe8251dc000
mmap(0x7fe8251df000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x11000) = 0x7fe8251df000
close(3)                                = 0
mprotect(0x7fe8251df000, 4096, PROT_READ) = 0
mprotect(0x7fe825251000, 4096, PROT_READ) = 0
mprotect(0x419000, 4096, PROT_READ)     = 0
brk(NULL)                               = 0x2249000
brk(0x224a000)                          = 0x224a000
brk(0x224f000)                          = 0x224f000
rt_sigprocmask(SIG_BLOCK, [HUP INT USR1 USR2 TERM], [], 8) = 0
geteuid()                               = 0
chdir("/")                              = 0
open("/dev/null", O_RDWR)               = 3
dup2(3, 0)                              = 0
dup2(3, 1)                              = 1
dup2(3, 2)                              = 2
close(3)                                = 0
rt_sigprocmask(SIG_BLOCK, ~[], [HUP INT USR1 USR2 TERM], 8) = 0
fork()                                  = 20527
rt_sigprocmask(SIG_SETMASK, [HUP INT USR1 USR2 TERM], NULL, 8) = 0
exit_group(0)                           = ?
+++ exited with 0 +++
```

hnyman, 2021-07-10:
Looking at your full strace log, I am unsure where the error is, or if there is an actual error.

At the first glance, your strace commands seem to not use the "-f" parameter for irqbalance, so you see just the errorless end/exit of the launch thread (pid 5342 5343 in long strace) , while the forked daemon thread (5344) seems to continue and poll regularly.

hnyman, 2023-01-14:
Old, stale. 
(New irqbalance versions in all supported branches.)

closed 2023-01-14 by hnyman