openwrt/packages #15907: Concerning CI testing

kinafu, 2021-06-19:
Probably I'm missing out on something, but I could not find any information concerning testing the packages for the ARM `mvebu-cortexa9` target.
I took a look on the CI result, where I noted the absence of testing for that particular architecture after @neheb commited a change, that was about to solve an issue I cared about.

I would just like to ask, how the packages for that platform (probably the `mvebu-cortexa9` target is just one representative for lots of different ARM architectures?) get tested. In case there is no automated testing, that might be nice to have.  
Doing so, issues with broken installation of packages could be detected automatically in advance.  

Some information about how you are handling things right now would be nice, since I have no insights and I just took a brief look on the CI and could not find docs on why there is testing for some archs and not for others.

neheb, 2021-06-21:
@aparcar might have more information.

aparcar, 2021-06-22:
[Looks like we test your arch as well](https://github.com/openwrt/packages/blob/master/.github/workflows/multi-arch-test-build.yml#L19). Anything missing?

kinafu, 2021-06-27:
For `mvebu-cortexa9` it says `runtime_test` is off. https://github.com/openwrt/packages/blob/03ff71d5ba955ad2d3f23de30e526ab6452297d4/.github/workflows/multi-arch-test-build.yml#L20

And later the testing will get skipped (also see the line following the one referenced below).
https://github.com/openwrt/packages/blob/03ff71d5ba955ad2d3f23de30e526ab6452297d4/.github/workflows/multi-arch-test-build.yml#L121

jefferyto, 2022-05-11:
Run-time testing is enabled for armvirt-32 - is there a significant enough difference between armvirt-32 and mvebu-cortexa9 to have run-time testing enabled for both?

neheb, 2022-05-11:
I think it’s a difference between NEON being available.