openwrt/packages #15929: strongswan: wrong secrets configuration

CodeFetch, 2021-06-21:
Maintainer: @pprindeville @Thermi

The UCI configuration partially misuses the secrets section in `swanctl.conf`.
Swanctl uses directories in `/etc/swanctl` like `pubkeys`, `rsa`, `private` etc. to automatically load certificates, public and private keys.

The swanctl.init in the strongswan package on the other hand defines a secrets section wherein it defines paths for such keys like here: 
https://github.com/openwrt/packages/blob/ed28f6e195bfa7bab7cfa226e13a8535264dc454/net/strongswan/files/swanctl.init#L483

These definitions are not intended to define "additional" files outside the `/etc/swanctl` sub folders, but are used for defining passphrases for encrypted keys according to the Strongswan documentation (rsa in the example above corresponds to the `/etc/swanctl/rsa` folder):
https://wiki.strongswan.org/projects/strongswan/wiki/Swanctlconf#secretsrsaltsuffixgt-section

Defining a secrets file here outside of /etc/swanctl/{`pubkeys`, `rsa`, `private` etc.} will NOT load it. So this is broken.

By the way one can put an RSA private key file in `/etc/swanctl/private` and from my understanding the `/etc/swanctl/rsa` folder exists for backwards compatibility only.

pprindeville, 2021-06-22:
I'm not sure I get the problem.

If you're using `gencerts` to generate your credentials, then they're being deposited in `/etc/swanctl/private/` per here:
https://github.com/openwrt/packages/blob/ed28f6e195bfa7bab7cfa226e13a8535264dc454/net/strongswan/files/gencerts.sh#L85

CodeFetch, 2021-06-22:
@pprindeville The secrets section is not for defining secret files to use for that connection, but for defining passphrases in case they are encrypted. swanctl automatically loads all secrets in `$SWANCTL_DIR/private/`. 
```
secrets.private<suffix>.file
File name in the private folder for which this passphrase should be used.
```
This option only makes sense for encrypted keys...

pprindeville, 2021-06-28:
Understood.

Fixed by PR #15973.

closed 2021-06-28 by pprindeville