openwrt/packages #15942: PyPi malicious packages

neheb, 2021-06-24:
https://arstechnica.com/gadgets/2021/06/counterfeit-pypi-packages-with-5000-downloads-installed-cryptominers/

Should we continue using PyPi?

Maintainer: @commodo @jefferyto

jefferyto, 2021-06-24:
Most (all?) of the attacks against PyPI have been typosquatting, where the attacker uploads a malicious package with a misspelled name (e.g. `maratlib` instead of `matplotlib`) and relies on people typing the wrong name when installing.

I believe an attacker can do the same thing on GitHub - pull from a legitimate repo, then push to GitHub under a misspelled user/organization name and correct repo name.

Checking the PyPI name or download URL is one of the things we do when we review packages (at least that is what I do and I hope/assume other reviewers are doing the same), so we should be catching these typos. I see no reason to not continue using PyPI.

neheb, 2021-06-24:
Alright.

closed 2021-06-24 by neheb