openwrt/packages #15951: mwan3  connected_v6_temp is full

zhihuiyuze, 2021-06-25:
Hi,i'm runing mwan3 (2.8.16-1) on openwrt 19.07.06.
I found that the console has an output
`mwan3 _connected_v6_temp is full maxelem 65536 reached`
After investigation,i think it is caused by limition of IPV6 Routing Table.
The routing table is imported init kernal by bird2,So how to remove these restrictions? Is this the problem?
Here's the configuration file
```
root@home-1-OpenWrt:/etc/config# cat mwan3

config rule 'vpn_ca_udp'
        option dest_ip '104.218.*.*
        option dest_port '52471'
        option proto 'udp'
        option sticky '0'
        option use_policy 'balanced'

config rule 'vpn_ca'
        option dest_ip '104.218.*.*'
        option dest_port '52471'
        option proto 'tcp'
        option sticky '0'
        option use_policy 'balanced'

config rule 'vpn_hk'
        option dest_ip '103.152.*.*'
        option sticky '0'
        option proto 'all'
        option use_policy 'balanced'

config rule 'heipv6_update'
        option dest_ip '64.62.200.2'
        option proto 'all'
        option sticky '0'
        option use_policy 'wan_only'

config rule 'direct'
        option proto 'all'
        option sticky '0'
        option ipset 'direct'
        option use_policy 'balanced'

config rule 'https_china'
        option dest_port '443'
        option ipset 'china'
        option sticky '1'
        option proto 'tcp'
        option use_policy 'balanced'

config rule 'China_v4'
        option proto 'all'
        option sticky '0'
        option ipset 'china'
        option use_policy 'balanced'

config rule 'ssh'
        option dest_port '22'
        option proto 'tcp'
        option sticky '1'
        option use_policy 'balanced'

config rule 'https'
        option dest_port '443'
        option proto 'tcp'
        option sticky '1'
        option use_policy 'VPN'

config rule 'ipv4_tcp'
        option sticky '0'
        option proto 'tcp'
        option dest_port '1024:65535'
[... 442 lines cut ...]
        };
    };
}

template bgp tpl_bgp_rs {
    graceful restart on;
    rs client;
    local as LOCAL_ASN;
    ipv6 {
        next hop self;
        import filter peer_in_v6;
        export filter{
            if bgp_export_all() then accept;
            if net ~ OUR_PREFIXES then bgp_path.prepend(141011);
        };
    };
}

protocol bgp home_1_arch from tpl_ibgp {  
    description "HOME_1_AECH BGP";
    source address 2602:feda:ab3::1;
    neighbor 2602:feda:ab3:0:3843:aeff:fe5b:18cb%'eth0' as 141011;
}

```

zhihuiyuze, 2021-06-26:
I try following [Preventing IPv6 rules](https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#preventing_ipv6_rules) to disable mwan3 ipv6 support,but it did't works.

feckert, 2021-06-29:
> I try following [Preventing IPv6 rules](https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#preventing_ipv6_rules) to disable mwan3 ipv6 support,but it did't works.

This is only valid for the mwan3 which is in the master or in the openwrt-21.02.

For the mwan3 in openwrt-19.07 you can change [this line](https://github.com/openwrt/packages/blob/openwrt-19.07/net/mwan3/files/lib/mwan3/mwan3.sh#L37) so that no IPV6 rules are created.

I can't tell you more, because I don't know bird2.

zhihuiyuze, 2021-07-14:
Hi,I have been upgrade to  OpenWrt 21.02-SNAPSHOT r16235-8921e36ed8 
After updating to this version, mwan3 cannot create routes normally.
I created a Wireguard rule so that all IPs from mainland China pass Wireguard. The configuration is the same as that of openwrt19, but it doesn’t work properly after 20.02.

```
Wed Jul 14 20:54:25 2021 user.info mwan3track[15717]: Check (ping) failed for target "1.1.1.1" on interface hk_1_vpn (hk_1_vpn). Current score: 6
Wed Jul 14 20:54:25 2021 user.notice mwan3track[15717]: Interface hk_1_vpn (hk_1_vpn) is disconnecting
Wed Jul 14 20:54:30 2021 user.notice mwan3track[15717]: Interface hk_1_vpn (hk_1_vpn) is connecting
Wed Jul 14 20:54:31 2021 user.info mwan3track[15717]: Lost 1 ping(s) on interface hk_1_vpn (hk_1_vpn). Current score: 5
```

But i can ping 1.1.1.1.
```
PING 1.1.1.1 (1.1.1.1): 56 data bytes
64 bytes from 1.1.1.1: seq=0 ttl=61 time=53.439 ms
64 bytes from 1.1.1.1: seq=1 ttl=61 time=51.358 ms
64 bytes from 1.1.1.1: seq=2 ttl=61 time=48.282 ms
64 bytes from 1.1.1.1: seq=3 ttl=61 time=48.070 ms
```
```
root@home-1-OpenWrt:/etc/rc.d# ip -4 route
default via 183.156.52.1 dev pppoe-wan proto static metric 120 
66.220.18.42 via 183.156.52.1 dev pppoe-wan proto static metric 120 
183.156.52.1 dev pppoe-wan proto kernel scope link src 183.156.52.190 
192.168.7.0/24 dev eth0 proto kernel scope link src 192.168.7.1 

```

```
Software-Version
-------------------------------------------------
OpenWrt - OpenWrt 21.02-SNAPSHOT r16235-8921e36ed8
LuCI - git-21.191.52202-07063ae

Output of "ip a show"
-------------------------------------------------
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 42:e0:77:ff:96:7c brd ff:ff:ff:ff:ff:ff
    inet 192.168.7.1/24 brd 192.168.7.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 240e:390:80c:3750::1/64 scope global dynamic noprefixroute 
       valid_lft 257545sec preferred_lft 171145sec
    inet6 2001:470:f050::1/64 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 2602:feda:ab3::1/64 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::40e0:77ff:feff:967c/64 scope link 
       valid_lft forever preferred_lft forever
3: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd ::
4: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
5: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
6: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
7: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1464 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
8: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd ::
9: siit0: <BROADCAST,MULTICAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 10
    link/ether 0a:95:f9:e3:42:55 brd ff:ff:ff:ff:ff:ff
10: Unmetered_VPN: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1280 qdisc noqueue state UNKNOWN group default qlen 1000
    link/none 
[... 442 lines cut ...]
        option interface 'hk_1_vpn'
        option metric '50'
        option weight '50'

config member 'hk_1_vpn_m60_w50'
        option interface 'hk_1_vpn'
        option metric '60'
        option weight '55'

config member 'wan1_m50_w50'
        option interface 'wan'
        option metric '50'
        option weight '50'

config member 'wan2_m50_w50'
        option metric '50'
        option weight '50'
        option interface 'wan2'

config member 'wan3_m50_w50'
        option interface 'wan3'
        option metric '50'
        option weight '50'

```

zhihuiyuze, 2021-07-18:
> Hi,I have been upgrade to OpenWrt 21.02-SNAPSHOT r16235-8921e36ed8
> After updating to this version, mwan3 cannot create routes normally.
> I created a Wireguard rule so that all IPs from mainland China pass Wireguard. The configuration is the same as that of openwrt19, but it doesn’t work properly after 20.02.
> 
> ```
> Wed Jul 14 20:54:25 2021 user.info mwan3track[15717]: Check (ping) failed for target "1.1.1.1" on interface hk_1_vpn (hk_1_vpn). Current score: 6
> Wed Jul 14 20:54:25 2021 user.notice mwan3track[15717]: Interface hk_1_vpn (hk_1_vpn) is disconnecting
> Wed Jul 14 20:54:30 2021 user.notice mwan3track[15717]: Interface hk_1_vpn (hk_1_vpn) is connecting
> Wed Jul 14 20:54:31 2021 user.info mwan3track[15717]: Lost 1 ping(s) on interface hk_1_vpn (hk_1_vpn). Current score: 5
> ```
> 
> But i can ping 1.1.1.1. on hk_1_vpn
> 
> ```
> PING 1.1.1.1 (1.1.1.1): 56 data bytes
> 64 bytes from 1.1.1.1: seq=0 ttl=61 time=53.439 ms
> 64 bytes from 1.1.1.1: seq=1 ttl=61 time=51.358 ms
> 64 bytes from 1.1.1.1: seq=2 ttl=61 time=48.282 ms
> 64 bytes from 1.1.1.1: seq=3 ttl=61 time=48.070 ms
> ```
> 
> ```
> Interface status:
>  interface wan is online 00h:30m:17s, uptime 00h:31m:29s and tracking is active
>  interface wan2 is offline and tracking is paused
>  interface wan3 is offline and tracking is paused
>  interface Unmetered_VPN is error (16) and tracking is active
>  interface Unmetered_VPN_2 is error (24) and tracking is active
>  interface Unmetered_VPN_3 is error (24) and tracking is active
>  interface hk_1_vpn is error (16) and tracking is active
> ```
> 
> It’s not online here, but it’s online again when shown here
> 
> [image]
> 
> ```
> root@home-1-OpenWrt:/etc/rc.d# ip -4 route
> default via 183.156.52.1 dev pppoe-wan proto static metric 120 
> 66.220.18.42 via 183.156.52.1 dev pppoe-wan proto static metric 120 
> 183.156.52.1 dev pppoe-wan proto kernel scope link src 183.156.52.190 
> 192.168.7.0/24 dev eth0 proto kernel scope link src 192.168.7.1 
> ```
> 
> ```
> Software-Version
> -------------------------------------------------
> OpenWrt - OpenWrt 21.02-SNAPSHOT r16235-8921e36ed8
> LuCI - git-21.191.52202-07063ae
> 
> 
> Output of "ip a show"
> -------------------------------------------------
> 1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
>     link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
>     inet 127.0.0.1/8 scope host lo
>        valid_lft forever preferred_lft forever
>     inet6 ::1/128 scope host 
>        valid_lft forever preferred_lft forever
> 2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
>     link/ether 42:e0:77:ff:96:7c brd ff:ff:ff:ff:ff:ff
>     inet 192.168.7.1/24 brd 192.168.7.255 scope global eth0
>        valid_lft forever preferred_lft forever
>     inet6 240e:390:80c:3750::1/64 scope global dynamic noprefixroute 
>        valid_lft 257545sec preferred_lft 171145sec
>     inet6 2001:470:f050::1/64 scope global noprefixroute 
>        valid_lft forever preferred_lft forever
>     inet6 2602:feda:ab3::1/64 scope global noprefixroute 
>        valid_lft forever preferred_lft forever
>     inet6 fe80::40e0:77ff:feff:967c/64 scope link 
[... 465 lines cut ...]
>         option metric '50'
>         option weight '50'
> 
> config member 'hk_1_vpn_m60_w50'
>         option interface 'hk_1_vpn'
>         option metric '60'
>         option weight '55'
> 
> config member 'wan1_m50_w50'
>         option interface 'wan'
>         option metric '50'
>         option weight '50'
> 
> config member 'wan2_m50_w50'
>         option metric '50'
>         option weight '50'
>         option interface 'wan2'
> 
> config member 'wan3_m50_w50'
>         option interface 'wan3'
>         option metric '50'
>         option weight '50'
> ```

This problem has been solved. It seems that after openwrt is updated, you need to turn on the add route in iwiregard and allow 0.0.0/0 to open at the same time to use the default gateway

closed 2021-07-22 by zhihuiyuze