openwrt/packages #15953: ocserv-1.1.3-1: requires non-existing libatomic1 package

mmokrejs, 2021-06-25:
```
Details for package ocserv

    Version: 1.1.3-1
    Size: ~234.8 KB installed
    Dependencies:
        libhttp-parser (14.7 KB) Not installed
        libgnutls (912.6 KB) Not installed
            libnettle8 (306.1 KB) Not installed
                libgmp10 (185.3 KB) Not installed
            libatomic1 Not available
        certtool (130.9 KB) Not installed
        libncurses6 Installed
            terminfo Installed
        libreadline8 Installed
            libncurses6 Installed
        libprotobuf-c (8.9 KB) Not installed
        libev (21.2 KB) Not installed
        kmod-tun Installed
            kernel Installed

Description

OpenConnect server (ocserv) is an SSL VPN server. Its purpose is to be a secure, small, fast and configurable VPN server. It implements the OpenConnect SSL VPN protocol, and has also (currently experimental) compatibility with clients using the AnyConnect SSL VPN protocol. The OpenConnect VPN protocol uses the standard IETF security protocols such as TLS 1.2, and Datagram TLS to provide the secure VPN service.

    Required dependency package libatomic1 is not available in any repository.
```

Powered by LuCI openwrt-21.02 branch (git-21.163.64918-ba57ec5) / OpenWrt 21.02.0-rc3 r16172-2aba3e9784

mmokrejs, 2022-05-22:
I did not build the package, I just installed it from the repo. I do not know why the repo did not contain the required dependency.

nmav, 2023-09-10:
Closing for now. Please reopen if the issue is still there.

closed 2023-09-10 by nmav