openwrt/packages #15960: collectd missing packages

kpoman, 2021-06-26:
Maintainer: @feckert 
Environment: ath79, OpenWrt SNAPSHOT, r17036-18c24a29f9, TPLink Archer C6 v2

Description:

```
All collectd packages are missing since a couple snapshots to now. See below:
```
root@openwrt-c6v2:~# opkg install collectd
Unknown package 'collectd'.
Collected errors:
 * opkg_install_cmd: Cannot install package collectd.
root@openwrt-c6v2:~# opkg install luci-app-statistics
Unknown package 'luci-app-statistics'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency collectd for luci-app-statistics
 * pkg_hash_check_unresolved: cannot find dependency collectd-mod-rrdtool for luci-app-statistics
 * pkg_hash_check_unresolved: cannot find dependency collectd-mod-iwinfo for luci-app-statistics
 * pkg_hash_check_unresolved: cannot find dependency collectd-mod-cpu for luci-app-statistics
 * pkg_hash_check_unresolved: cannot find dependency collectd-mod-memory for luci-app-statistics
 * pkg_hash_check_unresolved: cannot find dependency collectd-mod-interface for luci-app-statistics
 * pkg_hash_check_unresolved: cannot find dependency collectd-mod-load for luci-app-statistics
 * pkg_hash_check_unresolved: cannot find dependency collectd-mod-network for luci-app-statistics
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-statistics found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-statistics.
root@openwrt-c6v2:~# 
```
```

micmac1, 2021-06-26:
I guess this should sort itself out since fd13c12.

kpoman, 2021-06-28:
Great, apparently fixed, packages appeared again as built ! Thanks !

closed 2021-06-28 by kpoman