openwrt/packages #15986: node failed to build

kpoman, 2021-06-28:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @nxhack 
Environment: bcm27xx openwrt snapshot sdk

Description:
Failed to build node v14.17.1 with below error:
```
  cc -o /home/kpoman/Dev/openwrt/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64/build_dir/hostpkg/node-v14.17.1/out/Release/obj.target/openssl/deps/openssl/config/archs/linux-x86_64/asm/engines/e_padlock-x86_64.o ../deps/openssl/config/archs/linux-x86_64/asm/engines/e_padlock-x86_64.s '-DV8_DEPRECATION_WARNINGS' '-DV8_IMMINENT_DEPRECATION_WARNINGS' '-D__STDC_FORMAT_MACROS' '-DOPENSSL_NO_PINSHARED' '-DOPENSSL_THREADS' '-DOPENSSL_NO_HW' '-DNDEBUG' '-DOPENSSL_USE_NODELETE' '-DL_ENDIAN' '-DOPENSSL_PIC' '-DOPENSSL_CPUID_OBJ' '-DOPENSSL_IA32_SSE2' '-DOPENSSL_BN_ASM_MONT' '-DOPENSSL_BN_ASM_MONT5' '-DOPENSSL_BN_ASM_GF2m' '-DSHA1_ASM' '-DSHA256_ASM' '-DSHA512_ASM' '-DKECCAK1600_ASM' '-DRC4_ASM' '-DMD5_ASM' '-DAESNI_ASM' '-DVPAES_ASM' '-DGHASH_ASM' '-DECP_NISTZ256_ASM' '-DX25519_ASM' '-DPOLY1305_ASM' '-DOPENSSLDIR="/etc/ssl"' '-DENGINESDIR="/dev/null"' '-DTERMIOS' -I../deps/openssl/openssl -I../deps/openssl/openssl/include -I../deps/openssl/openssl/crypto -I../deps/openssl/openssl/crypto/include -I../deps/openssl/openssl/crypto/modes -I../deps/openssl/openssl/crypto/ec/curve448 -I../deps/openssl/openssl/crypto/ec/curve448/arch_32 -I../deps/openssl/config -I../deps/openssl/config/archs/linux-x86_64/asm -I../deps/openssl/config/archs/linux-x86_64/asm/include -I../deps/openssl/config/archs/linux-x86_64/asm/crypto -I../deps/openssl/config/archs/linux-x86_64/asm/crypto/include/internal  -pthread -Wall -Wextra -Wno-unused-parameter -m64 -Wa,--noexecstack -Wall -O3 -pthread -m64 -Wall -O3 -Wno-missing-field-initializers -Wno-old-style-declaration -O3 -fno-omit-frame-pointer  -MMD -MF /home/kpoman/Dev/openwrt/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64/build_dir/hostpkg/node-v14.17.1/out/Release/.deps//home/kpoman/Dev/openwrt/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64/build_dir/hostpkg/node-v14.17.1/out/Release/obj.target/openssl/deps/openssl/config/archs/linux-x86_64/asm/engines/e_padlock-x86_64.o.d.raw -I/home/kpoman/Dev/openwrt/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64/staging_dir/host/include -I/home/kpoman/Dev/openwrt/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64/staging_dir/hostpkg/include -I/home/kpoman/Dev/openwrt/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64/staging_dir/target-aarch64_cortex-a72_musl/host/include -O2 -I/home/kpoman/Dev/openwrt/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64/staging_dir/host/include -I/home/kpoman/Dev/openwrt/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64/staging_dir/hostpkg/include -I/home/kpoman/Dev/openwrt/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64/staging_dir/target-aarch64_cortex-a72_musl/host/include -c
make[4]: execvp: printf: Argument list too long
make[4]: *** [deps/openssl/openssl.target.mk:905: /home/kpoman/Dev/openwrt/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64/build_dir/hostpkg/node-v14.17.1/out/Release/obj.target/deps/openssl/libopenssl.a] Error 127
make[3]: *** [Makefile:104: node] Error 2
make[3]: Leaving directory '/home/kpoman/Dev/openwrt/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64/build_dir/hostpkg/node-v14.17.1'
make[2]: *** [Makefile:157: /home/kpoman/Dev/openwrt/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64/build_dir/hostpkg/node-v14.17.1/.built] Error 2
make[2]: Leaving directory '/home/kpoman/Dev/openwrt/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64/feeds/packages/lang/node'
time: package/feeds/packages/node/host-compile#101.79#14.28#123.60
    ERROR: package/feeds/packages/node [host] failed to build.
make[1]: *** [package/Makefile:114: package/feeds/packages/node/host/compile] Error 1
make[1]: Leaving directory '/home/kpoman/Dev/openwrt/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64'
make: *** [/home/kpoman/Dev/openwrt/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64/include/toplevel.mk:223: package/feeds/packages/ntopng/compile] Error 2
```
error make[4]: execvp: printf: Argument list too long, apparently fixed [here](https://github.com/nodejs/node/issues/9137)

nxhack, 2021-06-28:
Hi @kpoman 

Patch is available. 

https://raw.githubusercontent.com/nxhack/openwrt-node-packages/master/node/patches/v14.x/010-execvp-arg-list-too-long.patch

However, I do not plan to apply it to this community package.

nxhack, 2021-06-29:
A simple solution is to rename the PATH name, as shown in this link.

https://github.com/nodejs/node/issues/9137#issuecomment-370728059

kpoman, 2021-06-29:
> Hi @kpoman
> 
> Patch is available.
> 
> https://raw.githubusercontent.com/nxhack/openwrt-node-packages/master/node/patches/v14.x/010-execvp-arg-list-too-long.patch
> 
> However, I do not plan to apply it to this community package.

Hello @nxhack ! Thanks for pointing that patch ! Why dont you want to commit it to the community package (just for curiosity) ?

nxhack, 2021-06-29:
@kpoman 

As discussed in the upstream,

* It depends on the situation of the OS (kernel) you are building on.
* Easy to work around by "rename".
* The patch to fix this is not smart.