openwrt/packages #15988: python-dateutil: Compilation Fails on Debian 11

DeathCamel58, 2021-06-28:
# Issue template (remove lines from top till here)

Maintainer: @commodo (find it by checking history of the package Makefile)
Environment: 
```
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```

Description:
`python-dateutil` fails to compile with setuptools version 6.0.1 (which the build system uses). It looks like support was dropped for Python 2. 
It also looks like the `19.07` branch buildbot failed to compile this package as well.

Did some research and found these:
[Very similar setuptools error](https://github.com/pypa/setuptools_scm/issues/541)

[Failing build log from buildbot](https://downloads.openwrt.org/releases/faillogs-19.07/aarch64_generic/packages/python-dateutil/python/compile.txt)

commodo, 2021-06-29:
> Maintainer: @commodo (find it by checking history of the package Makefile)
> Environment:
>
> ```
> No LSB modules are available.
> Distributor ID:	Debian
> Description:	Debian GNU/Linux 11 (bullseye)
> Release: [...]

so, this looks like an issue on 19.07 only
will check

commodo, 2021-06-29:
this seems to fix it  https://github.com/openwrt/packages/pull/15995

DeathCamel58, 2021-06-29:
Yep, thanks!

closed 2021-06-29 by DeathCamel58