openwrt/packages #16001: mosquitto-nossl has missing dependencies: librt [OpenWrt 21.02.0-rc3]

MaciejMilinkiewicz, 2021-06-30:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)

Environment: 
Model | TP-Link Archer C20 v4
Architecture | MediaTek MT7628AN ver:1 eco:2

Model	Xiaomi Mi Router 4A Gigabit Edition
Architecture	MediaTek MT7621 ver:1 eco:3

Description:
Trying to install mosquitto-nossl package I am getting the following warning:
**librt Not available**

```
Details for package mosquitto-nossl

    Version: 2.0.11-1
    Size: ~85.4 KB installed
    Dependencies:
        librt Not available

Description

Mosquitto is an open source (BSD licensed) message broker that implements the MQTT protocol version 3.1, 3.1.1 and 5. MQTT provides a lightweight method of carrying out messaging using a publish/subscribe model. This package also includes some basic support for configuring via UCI This package is built WITHOUT SSL support.

    Required dependency package librt is not available in any repository.
```

karlp, 2021-06-30:
Have you done an opkg update?  This doesn't _appear_ to be a mosquitto problem at least?

MaciejMilinkiewicz, 2021-06-30:
I've done opkg update.

I've checked also my:
Model	TP-Link Archer D7 v1
Architecture	Qualcomm Atheros QCA9558 ver 1 rev 0
Firmware Version	OpenWrt 21.02.0-rc3 r16172-2aba3e9784 / LuCI openwrt-21.02 branch git-21.163.64918-ba57ec5

and there is librt, so I find it:
https://downloads.openwrt.org/releases/21.02.0-rc3/targets/ath79/generic/packages/librt_1.1.24-3_mips_24kc.ipk
but for MT7628AN and MT7621 also are there:
https://downloads.openwrt.org/releases/21.02.0-rc3/targets/ramips/mt76x8/packages/librt_1.1.24-3_mipsel_24kc.ipk
https://downloads.openwrt.org/releases/21.02.0-rc3/targets/ramips/mt7621/packages/librt_1.1.24-3_mipsel_24kc.ipk
but some reasons opkg can't use/find them.

I've download them, install manually and they work.
Probably the problem is in opkg.

champtar, 2021-08-20:
similar to #15560

hnyman, 2023-01-14:
Old. Based on comments in #15560, likely a temporary LuCI 21.02 issue that has been fixed.  
(And in any case, something either in the core opkg or in LuCI)

closed 2023-01-14 by hnyman