openwrt/packages #16002: tessdata: takes forever to build because of single-threaded steps

zorun, 2021-06-30:
Maintainer: @vk496 @neheb 

Description:

tessdata takes forever to build, even on a multi-core machine with -jXX.

Looking at the build, there are lots of single-threaded steps doing `gzip -d` / `gzip` one after the other. Each gzip takes about 5 seconds, and the total is several minutes.

I suspected this is related to the `foreach LANG,$(ALLTESSERACTLANG)` that builds sequentially.

neheb, 2021-06-30:
Most likely. Patches welcome.

zorun, 2021-06-30:
Right, the main issue is that we uncompress 130 times the same big tarball... (638 MB)

I didn't find how to parallelize package creation, but fixing this uncompress issue already improved things a lot.

PR here: #16003

vk496, 2021-06-30:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAFQAAABUCAYAAAAcaxDBAAAAAXNSR0IArs4c6QAAAARzQklUCAgICHwIZIgAAAAySURBVHic7cEBDQAAAMKg909tDjegAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAeDVulAABbzDScQAAAABJRU5ErkJggg==###24x24:true###][Imagen de rastreo][https://github.com/notifications/beacon/ABJ7L2YDR6ECPFUGJ6U6SM3TVMDVBA5CNFSM47R366N2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOGPX46WI.gif]

Thank you!

It was the first time I created a complex package in OpenWrt repos.

closed 2021-07-01 by neheb

zorun, 2021-07-02:
On the buildbots, before the change: `time: package/feeds/packages/tessdata/compile#4607.45#481.64#5370.20`

After: `time: package/feeds/packages/tessdata/compile#179.02#35.19#343.18`

This is user/system/realtime, so we went from 5370s to 343s, not bad :)