openwrt/packages #16007: Luci-app-nft-qos: Not functioning anymore since r16068

Thebroughfamily, 2021-06-30:
Description:

Nft-qos package no longer functions on latest 21.02 snapshot builds, last working build was r16068-4b89b90. Can set limits on the app as normal, no error messages appear but settings have no effect. Timing of issue appears related to the change in the interfaces tab.

wulfy23, 2021-07-16:
-pinging the maintainer @rosysong-
-including your (sanitized) configs (describing your use case in detail)
-linking your forum thread

would likely assist your cause considerably...

Thebroughfamily, 2021-07-16:
I would if I knew how to do all this stuff, I'm a very basic user of the software, I have no cli knowledge.

Thebroughfamily, 2021-07-25:
@rosysong 

Will nft-qos be modified for the latest version of openwrt?

> Problem is the underlying nft-qos scripts being broken since the netifd ifname->device migration

champtar, 2021-08-12:
@Thebroughfamily can you try to test https://github.com/openwrt/packages/pull/16350 ?

professor-jonny, 2021-12-09:
I doubt you will get in contact with the maintainer their company closed in june 2019 their domain has closed and their github email is addressed there.

They were manufacturing openwrt devices much like wallystech.

wulfy23, 2022-01-31:
tested mac-limit yesterday and worked perfectly ok

this can be closed imho and thankyou @champtar for the fix...

wonson, 2023-01-17:
I can reproduce it on my router 

D-Link DIR-867 A1
MediaTek MT7621 ver:1 eco:3
OpenWrt 21.02.5 r16688-fa9a932fdb / LuCI openwrt-21.02 branch git-22.347.45520-d30ab74

wehagy, 2025-09-25:
Package removed in the commit https://github.com/openwrt/packages/commit/14bff6e90ef8ebe22d72b4ede398e306ea3b46da.

closed 2025-09-25 by wehagy