openwrt/packages #16009: syncthing: looking for maintainer

aparcar, 2021-07-01:
Hi, some time ago I added the package syncthing to OpenWrt but I don't have the time to maintain it. Is anyone using is at excited to keep it up to date?

pmoya-in-the-web, 2021-12-16:
Hi, 
i am really new to openwrt and syncthing. Just bought a Linksys WR32X router few weeks ago and I am still configuring it with openwrt & syncthing. 

Maybe I could help you with documentation and some easy tasks by now.

jefferyto, 2023-12-09:
@brvphoenix you mentioned in https://github.com/openwrt/packages/pull/19695#issuecomment-1302958185 that you are okay with becoming the maintainer - is that still the case? If so, please update the `PKG_MAINTAINER` value in the package makefile with your details.

closed 2023-12-18 by hnyman