openwrt/packages #16045: [sqm-scripts/luci-app-sqm] RTT (target latency) value for cake in sqm-scripts is getting ignored.

AreYouLoco, 2021-07-05:
Hi,

I am testing sqm-scripts capability with my router to get rid of bufferbloat. So far it works well but I wan to fine tune it to the edge. So I stumped on a bug that one of the LUCI/config settings is getting ignored by sqm-scripts.

Here is what I mean (Here is my full /etc/config/sqm):

> config queue 'eth1'
>         option interface 'eth2'
>         option debug_logging '1'
>         option verbosity '5'
>         option qdisc 'cake'
>         option enabled '1'
>         option upload '5415'
>         option qdisc_advanced '1'
>         option squash_dscp '1'
>         option ingress_ecn 'ECN'
>         option qdisc_really_really_advanced '1'
>         option squash_ingress '0'
>         option egress_ecn 'ECN'
>         option linklayer 'ethernet'
>         option overhead '38'
>         option linklayer_advanced '1'
>         option tcMTU '1538'
>         option tcTSIZE '128'
>         option linklayer_adaptation_mechanism 'cake'
>         option iqdisc_opts 'nat dual-dsthost ingress diffserv4'
>         option eqdisc_opts 'nat dual-srchost ack-filter diffserv4'
>         option script 'layer_cake.qos'
>         option tcMPU '84'
>         option download '27598'
**>         option itarget '33ms'**
**>         option etarget '33ms'**

This **suppose to set rtt option in qdisc queue to 33ms but it stays default at 100ms.** There must be a problem with parsing the config somewhere. I tried /etc/init.d/sqm restart and full device reboot the setting is not being taken into account. This is how I test it:

> root@router:~# tc -s qdisc show dev eth2 | grep qdisc
> qdisc cake 8031: root refcnt 9 bandwidth 5415Kbit diffserv4 dual-srchost nat nowash ack-filter split-gso **rtt 100.0ms** noatm overhead 38 mpu 84 
> qdisc ingress ffff: parent ffff:fff1 ----------------

This config should put there my setting from config which is **33ms**. I think those scripts are platform independent so not much info about my board needed.

AreYouLoco, 2021-07-05:
* DEBUG LOG:

> root@router:~# /etc/init.d/sqm start
> SQM: 
> SQM: Mon Jul  5 19:38:27 CEST 2021: Starting.
> SQM: Starting SQM script: layer_cake.qos on eth2, in: 27598 Kbps, out: 5415 Kbps
> SQM: fn_exists: function candidate name: sqm_start
> SQM: fn_exists: TYPE_OUTPUT: sqm_start: not found
> SQM: fn_exists: return value: 1
> SQM: Using generic sqm_start_default function.
> SQM: fn_exists: function candidate name: sqm_prepare_script
> SQM: fn_exists: TYPE_OUTPUT: sqm_prepare_script is a function
> SQM: fn_exists: return value: 0
> SQM: sqm_start_default: starting sqm_prepare_script
> SQM: cmd_wrapper: COMMAND: /sbin/ip link add name TMP_IFB_4_SQM type ifb
> SQM: cmd_wrapper: ip: SUCCESS: /sbin/ip link add name TMP_IFB_4_SQM type ifb
> SQM: cmd_wrapper: COMMAND: /usr/sbin/tc qdisc replace dev TMP_IFB_4_SQM root cake
> SQM: cmd_wrapper: tc: SUCCESS: /usr/sbin/tc qdisc replace dev TMP_IFB_4_SQM root cake
> SQM: QDISC cake is useable.
> SQM: cmd_wrapper: COMMAND: /sbin/ip link set dev TMP_IFB_4_SQM down
> SQM: cmd_wrapper: ip: SUCCESS: /sbin/ip link set dev TMP_IFB_4_SQM down
> SQM: cmd_wrapper: COMMAND: /sbin/ip link delete TMP_IFB_4_SQM type ifb
> SQM: cmd_wrapper: ip: SUCCESS: /sbin/ip link delete TMP_IFB_4_SQM type ifb
> SQM: cmd_wrapper: COMMAND: /sbin/ip link add name TMP_IFB_4_SQM type ifb
> SQM: cmd_wrapper: ip: SUCCESS: /sbin/ip link add name TMP_IFB_4_SQM type ifb
> SQM: cmd_wrapper: COMMAND: /usr/sbin/tc qdisc replace dev TMP_IFB_4_SQM root cake
> SQM: cmd_wrapper: tc: SUCCESS: /usr/sbin/tc qdisc replace dev TMP_IFB_4_SQM root cake
> SQM: QDISC cake is useable.
> SQM: cmd_wrapper: COMMAND: /sbin/ip link set dev TMP_IFB_4_SQM down
> SQM: cmd_wrapper: ip: SUCCESS: /sbin/ip link set dev TMP_IFB_4_SQM down
> SQM: cmd_wrapper: COMMAND: /sbin/ip link delete TMP_IFB_4_SQM type ifb
> SQM: cmd_wrapper: ip: SUCCESS: /sbin/ip link delete TMP_IFB_4_SQM type ifb
> SQM: sqm_start_default: Starting layer_cake.qos
> SQM: ifb associated with interface eth2: 
> SQM: Currently no ifb is associated with eth2, this is normal during starting of the sqm system.
> SQM: cmd_wrapper: COMMAND: /sbin/ip link add name ifb4eth2 type ifb
> SQM: cmd_wrapper: ip: SUCCESS: /sbin/ip link add name ifb4eth2 type ifb
> SQM: fn_exists: function candidate name: egress
> SQM: fn_exists: TYPE_OUTPUT: egress is a function
> SQM: fn_exists: return value: 0
> SQM: cmd_wrapper: tc: invocation silenced by request, FAILURE either expected or acceptable.
> SQM: cmd_wrapper: COMMAND: /usr/sbin/tc qdisc del dev eth2 root
> SQM: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev eth2 root
> SQM: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
> SQM: cake link layer adjustments:  overhead 38 mpu 84
> SQM: cmd_wrapper: COMMAND: /usr/sbin/tc qdisc add dev eth2 root cake bandwidth 5415kbit overhead 38 mpu 84 diffserv3 nat dual-srchost ack-filter diffserv4
> SQM: cmd_wrapper: tc: SUCCESS: /usr/sbin/tc qdisc add dev eth2 root cake bandwidth 5415kbit overhead 38 mpu 84 diffserv3 nat dual-srchost ack-filter diffserv4
> SQM: sqm_start_default: egress shaping activated
> SQM: cmd_wrapper: COMMAND: /sbin/ip link add name TMP_IFB_4_SQM type ifb
> SQM: cmd_wrapper: ip: SUCCESS: /sbin/ip link add name TMP_IFB_4_SQM type ifb
> SQM: cmd_wrapper: COMMAND: /usr/sbin/tc qdisc replace dev TMP_IFB_4_SQM ingress
> SQM: cmd_wrapper: tc: SUCCESS: /usr/sbin/tc qdisc replace dev TMP_IFB_4_SQM ingress
> SQM: QDISC ingress is useable.
> SQM: cmd_wrapper: COMMAND: /sbin/ip link set dev TMP_IFB_4_SQM down
> SQM: cmd_wrapper: ip: SUCCESS: /sbin/ip link set dev TMP_IFB_4_SQM down
> SQM: cmd_wrapper: COMMAND: /sbin/ip link delete TMP_IFB_4_SQM type ifb
> SQM: cmd_wrapper: ip: SUCCESS: /sbin/ip link delete TMP_IFB_4_SQM type ifb
> SQM: fn_exists: function candidate name: ingress
> SQM: fn_exists: TYPE_OUTPUT: ingress is a function
> SQM: fn_exists: return value: 0
> SQM: cmd_wrapper: tc: invocation silenced by request, FAILURE either expected or acceptable.
> SQM: cmd_wrapper: COMMAND: /usr/sbin/tc qdisc del dev eth2 handle ffff: ingress
> SQM: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev eth2 handle ffff: ingress
> SQM: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: Invalid argument
> SQM: cmd_wrapper: COMMAND: /usr/sbin/tc qdisc add dev eth2 handle ffff: ingress
> SQM: cmd_wrapper: tc: SUCCESS: /usr/sbin/tc qdisc add dev eth2 handle ffff: ingress
> SQM: cmd_wrapper: tc: invocation silenced by request, FAILURE either expected or acceptable.
> SQM: cmd_wrapper: COMMAND: /usr/sbin/tc qdisc del dev ifb4eth2 root
> SQM: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev ifb4eth2 root
> SQM: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
> SQM: cake link layer adjustments:  overhead 38 mpu 84
> SQM: cmd_wrapper: COMMAND: /usr/sbin/tc qdisc add dev ifb4eth2 root cake bandwidth 27598kbit overhead 38 mpu 84 diffserv3 wash nat dual-dsthost ingress diffserv4
> SQM: cmd_wrapper: tc: SUCCESS: /usr/sbin/tc qdisc add dev ifb4eth2 root cake bandwidth 27598kbit overhead 38 mpu 84 diffserv3 wash nat dual-dsthost ingress diffserv4
> SQM: cmd_wrapper: COMMAND: /sbin/ip link set dev ifb4eth2 up
> SQM: cmd_wrapper: ip: SUCCESS: /sbin/ip link set dev ifb4eth2 up
> SQM: cmd_wrapper: COMMAND: /usr/sbin/tc filter add dev eth2 parent ffff: protocol all prio 10 u32 match u32 0 0 flowid 1:1 action mirred egress redirect dev ifb4eth2
> SQM: cmd_wrapper: tc: SUCCESS: /usr/sbin/tc filter add dev eth2 parent ffff: protocol all prio 10 u32 match u32 0 0 flowid 1:1 action mirred egress redirect dev ifb4eth2
> SQM: sqm_start_default: ingress shaping activated
> SQM: layer_cake.qos was started on eth2 successfully

The log confirms that this setting is simply ignored. Manually setting that with this command...

> /usr/sbin/tc qdisc add dev eth2 root cake bandwidth 27598kbit overhead 38 mpu 84 diffserv3 wash nat dual-dsthost ingress diffserv4 rtt 33.ms

...simply works!

AreYouLoco, 2021-07-05:
I just spotted that also diffserv4 from config is changed when run from script into diffserv3.
So for sure there are some options parsing errors.

moeller0, 2021-07-05:
Cake does not allow to set the target from tc, only the interval and that is not exposed in the LuCI GUI at all.
Target is roughy the permissible standing queue duration, while interval is the time the delay needs to be above target to elicit the first drop/mark (this will also schedule the next drop/mark less than interval in the future).

moeller0, 2021-07-05:
> I just spotted that also diffserv4 from config is changed when run from script into diffserv3.
> So for sure there are some options parsing errors.

I believe this is not a bug, we set diffserv3 as default in the scripts and when you specify:
option eqdisc_opts 'nat dual-srchost ack-filter diffserv4'
in your config, that diffserv4 will get added to the end of the cake invocation:
/usr/sbin/tc qdisc add dev eth2 root cake bandwidth 5415kbit overhead 38 mpu 84 diffserv3 nat dual-srchost ack-filter diffserv4
and the resulting cake instance reports:
qdisc cake 8031: root refcnt 9 bandwidth 5415Kbit diffserv4 dual-srchost nat nowash ack-filter split-gso rtt 100.0ms noatm overhead 38 mpu 84
I might not be seeing the error here?

moeller0, 2021-07-05:
But yes, diffserv selection should ideally move into the GUI as explicit drop box or similar.

moeller0, 2021-07-05:
Ah, just tested it, I see the issue, the quick fix is to simply edit /usr/lib/sqm/defaults.sh:
[ -z "$INGRESS_CAKE_OPTS" ] && INGRESS_CAKE_OPTS="diffserv3"                                                                                                  
[ -z "$EGRESS_CAKE_OPTS" ] && EGRESS_CAKE_OPTS="diffserv3"

AreYouLoco, 2021-07-06:
What about rtt value? Is it target ingress/target egress in luci?

AreYouLoco, 2021-07-17:
I've successfully set rtt using your sugestion. Closing issue.

closed 2021-07-17 by AreYouLoco