openwrt/packages #16058: https-dns-proxy "https_client.c:205 No response" Spam

gilvela-mongodb, 2021-07-06:
Maintainer: @stangri  
Environment: x86-64 AM1M-S2H/Athlon 5350 - OpenWrt 19.07.1

Description:
After upgrading package to version 2021-06-03-1 I am getting this output in syslog just after starting the process:
```
Tue Jul  6 17:41:23 2021;10.122.122.1; <30>Jul  6 17:41:24 RouterVH https-dns-proxy[869]: [E] 1625611284.733891 https_client.c:117 CURLOPT_HTTP_VERSION error 1: Error
Tue Jul  6 17:41:23 2021;10.122.122.1; <30>Jul  6 17:41:24 RouterVH https-dns-proxy[869]: [E] 1625611284.733941 https_client.c:119 Try to run application with -x argument! Forcing HTTP/1.1 version.
Tue Jul  6 17:41:23 2021;10.122.122.1; <30>Jul  6 17:41:24 RouterVH https-dns-proxy[868]: [E] 1625611284.733916 https_client.c:117 CURLOPT_HTTP_VERSION error 1: Error
Tue Jul  6 17:41:23 2021;10.122.122.1; <30>Jul  6 17:41:24 RouterVH https-dns-proxy[868]: [E] 1625611284.733949 https_client.c:119 Try to run application with -x argument! Forcing HTTP/1.1 version.
```

Then I get almost a continuous spam:
```
Tue Jul  6 17:41:46 2021;10.122.122.1; <30>Jul  6 17:41:47 RouterVH https-dns-proxy[869]: [E] 1625611307.747611 https_client.c:205 No response
Tue Jul  6 17:41:48 2021;10.122.122.1; <30>Jul  6 17:41:49 RouterVH https-dns-proxy[869]: [E] 1625611309.640747 https_client.c:205 No response
Tue Jul  6 17:41:48 2021;10.122.122.1; <30>Jul  6 17:41:49 RouterVH https-dns-proxy[869]: [E] 1625611309.640786 https_client.c:205 No response
Tue Jul  6 17:41:48 2021;10.122.122.1; <30>Jul  6 17:41:49 RouterVH https-dns-proxy[869]: [E] 1625611309.640801 https_client.c:205 No response
Tue Jul  6 17:41:48 2021;10.122.122.1; <30>Jul  6 17:41:49 RouterVH https-dns-proxy[869]: [E] 1625611309.643522 https_client.c:205 No response
Tue Jul  6 17:41:48 2021;10.122.122.1; <30>Jul  6 17:41:49 RouterVH https-dns-proxy[869]: [E] 1625611309.643811 https_client.c:205 No response
Tue Jul  6 17:41:50 2021;10.122.122.1; <30>Jul  6 17:41:51 RouterVH https-dns-proxy[869]: [E] 1625611311.520540 https_client.c:205 No response
Tue Jul  6 17:41:50 2021;10.122.122.1; <30>Jul  6 17:41:51 RouterVH https-dns-proxy[869]: [E] 1625611311.520571 https_client.c:205 No response
Tue Jul  6 17:41:50 2021;10.122.122.1; <30>Jul  6 17:41:51 RouterVH https-dns-proxy[869]: [E] 1625611311.520586 https_client.c:205 No response
Tue Jul  6 17:41:50 2021;10.122.122.1; <30>Jul  6 17:41:51 RouterVH https-dns-proxy[869]: [E] 1625611311.521238 https_client.c:205 No response
Tue Jul  6 17:41:50 2021;10.122.122.1; <30>Jul  6 17:41:51 RouterVH https-dns-proxy[869]: [E] 1625611311.522787 https_client.c:205 No response
```

stangri, 2021-07-07:
Have you tried forcing HTTP/1 and see if it helps? Also, which resolvers are you using?

gilvela-mongodb, 2021-07-18:
It seems adding the use_http1 = 1 helped with the initial errors, however I am still seeing the spam of https_client.c:205 No response

I am using Cloudfare as first responder and OpenDNS as second one

here is the contents of my /etc/config/http-dns-proxy file:
```

config main 'config'
        option update_dnsmasq_config '*'
        option force_dns '1'
        list force_dns_port '53'
        list force_dns_port '853'

config https-dns-proxy
        option resolver_url 'https://cloudflare-dns.com/dns-query'
        option listen_addr '127.0.0.1'
        option user 'nobody'
        option group 'nogroup'
        option bootstrap_dns '1.1.1.1,1.0.0.1'
        option listen_port '5053'
        option use_http1 '1'

config https-dns-proxy
        option listen_addr '127.0.0.1'
        option user 'nobody'
        option group 'nogroup'
        option bootstrap_dns '208.67.222.222,208.67.220.220'
        option resolver_url 'https://doh.opendns.com/dns-query'
        option listen_port '5054'
        option use_http1 '1'

```

stangri, 2021-07-18:
Is your connection flakey?

jjlin, 2021-08-08:
I just upgraded to OpenWrt 19.07.8 and reinstalled https-dns-proxy, and now I have the same issue. It was working fine right before the upgrade, but most DNS resolution via https-dns-proxy seems to be broken now, so I'm just adding other DNS servers to my DNS forwardings for now. My logs look like this:

```
Sun Aug  8 02:55:15 2021 daemon.info https-dns-proxy[1346]: [F] 1628416515.055292 dns_server.c:45 Error binding 127.0.0.1:5053: Address not available (-1)
Sun Aug  8 02:55:15 2021 daemon.info https-dns-proxy[1347]: [F] 1628416515.056775 dns_server.c:45 Error binding 127.0.0.1:5054: Address not available (-1)
Sun Aug  8 02:55:20 2021 daemon.info https-dns-proxy[1744]: [E] 1628416520.581136 https_client.c:117 CURLOPT_HTTP_VERSION error 1: Error
Sun Aug  8 02:55:20 2021 daemon.info https-dns-proxy[1743]: [E] 1628416520.583595 https_client.c:117 CURLOPT_HTTP_VERSION error 1: Error
Sun Aug  8 02:55:20 2021 daemon.info https-dns-proxy[1743]: [E] 1628416520.583659 https_client.c:119 Try to run application with -x argument! Forcing HTTP/1.1 version.
Sun Aug  8 02:55:20 2021 daemon.info https-dns-proxy[1744]: [E] 1628416520.623955 https_client.c:119 Try to run application with -x argument! Forcing HTTP/1.1 version.
Sun Aug  8 03:05:39 2021 daemon.info https-dns-proxy[1743]: [E] 1628417139.872644 https_client.c:205 No response
Sun Aug  8 03:05:40 2021 daemon.info https-dns-proxy[1744]: [E] 1628417140.089580 https_client.c:205 No response
Sun Aug  8 03:05:44 2021 daemon.info https-dns-proxy[1744]: [E] 1628417144.389314 https_client.c:205 No response
Sun Aug  8 03:05:45 2021 daemon.info https-dns-proxy[1743]: [E] 1628417145.570447 https_client.c:205 No response
Sun Aug  8 03:15:17 2021 daemon.info https-dns-proxy[1743]: [E] 1628417717.520679 https_client.c:205 No response
Sun Aug  8 03:15:17 2021 daemon.info https-dns-proxy[1743]: [E] 1628417717.537453 https_client.c:205 No response
Sun Aug  8 03:15:18 2021 daemon.info https-dns-proxy[1744]: [E] 1628417718.418780 https_client.c:205 No response
Sun Aug  8 03:15:18 2021 daemon.info https-dns-proxy[1743]: [E] 1628417718.520027 https_client.c:207 curl response code: 500, content length: 0
Sun Aug  8 03:15:18 2021 daemon.info https-dns-proxy[1743]: [E] 1628417718.528390 https_client.c:207 curl response code: 500, content length: 0
```

There were basically continuous `No response` lines before I added the other DNS servers.

Tonkercke, 2021-08-08:
I also encountered the same problem ！

The following prompt is constantly displayed in the log ：

daemon.info https-dns-proxy[4471]: [E] 1628419666.257927 https_client.c:351 3A0D: CURLINFO_OS_ERRNO: 145 Operation timed out

jjlin, 2021-08-12:
After upgrading to `https-dns-proxy` 2021-07-29-1, which presumably contains aarond10/https_dns_proxy#124, I now get somewhat more helpful messages like
```
https-dns-proxy[7125]: [E] 1628727543.269371 https_client.c:297 DD21: Connecting and sending request to resolver was successful, but no response was sent back
https-dns-proxy[7125]: [E] 1628727543.270217 https_client.c:300 Resolver may not support current HTTP/1.1 protocol version
```
It looks like curl with HTTP/2 support (#16255) hasn't been released yet, but the DoH resolvers I'm using (Cloudflare and Google) seem to work fine with HTTP/1.1, so it's puzzling why this is happening. For example:
```sh
root@OpenWrt:~# curl --http1.1 -H 'accept: application/dns-json' 'https://cloudflare-dns.com/dns-query?name=dns.google'
{"Status":0,"TC":false,"RD":true,"RA":true,"AD":true,"CD":false,"Question":[{"name":"dns.google","type":1}],"Answer":[{"name":"dns.google","type":1,"TTL":406,"data":"8.8.8.8"},{"name":"dns.google","type":1,"TTL":406,"data":"8.8.4.4"}]}
```

stangri, 2021-08-18:
@jjlin curl with HTTP/2 support has been merged a few weeks ago, are you still getting the errors?

jjlin, 2021-08-18:
@stangri Yes. #16255 was merged, but doesn't seem to have been released. https://openwrt.org/packages/pkgdata/libcurl4 still shows 7.66.0-3, which is what I have installed.

jjlin, 2021-08-18:
Also, I just tried reverting back to OpenWrt 19.07.7, and the problem still exists with the current version of `https-dns-proxy` (I don't know how to go back to the old version that was in place when 19.07.7 was first released). So the problem was likely introduced in a release of the `https-dns-proxy` package between 19.07.7 and 19.07.8.

idarek, 2021-08-29:
Same issue here, unable to use `https-dns-proxy` at all, as DNS resolve is very slow and hanging over night. 

> Issue [16486](https://github.com/openwrt/packages/issues/16486) may be merged with this one.

SmallAndSoft, 2021-09-19:
I am affected by DoH stopping working and spamming those:
 `Sun Sep 19 09:28:35 2021 daemon.info https-dns-proxy[9571]: [E] 1632043715.723151 https_client.c:297 7A34: Connecting and sending request to resolver was successful, but no response was sent back
Sun Sep 19 09:28:35 2021 daemon.info https-dns-proxy[9571]: [E] 1632043715.724531 https_client.c:300 Resolver may not support current HTTP/1.1 protocol version
`

idarek, 2021-09-19:
In my case updating to stable 21.02 helped

stangri, 2021-09-19:
> I am affected by DoH stopping working and spamming those:
> `Sun Sep 19 09:28:35 2021 daemon.info https-dns-proxy[9571]: [E] 1632043715.723151 https_client.c:297 7A34: Connecting and sending request to resolver was successful, but no response was sent back Sun Sep 19 09:28:35 2021 daemon.info https-dns-proxy[9571]: [E] 1632043715.724531 https_client.c:300 Resolver may not support current HTTP/1.1 protocol version `
> All while ping 1.1.1.1 works fine. Restarting helps but it is annoying because of how random it happens.

You may want to change resolvers or switch to a newer OpenWrt.

stangri, 2021-10-01:
This issue has been fixed in OpenWrt 21.02.0, but the fix is not going to be backported to 19.07. Please upgrade to OpenWrt 21.02.0 or use different (not requiring HTTP/2) resolvers.

closed 2021-10-01 by stangri