openwrt/packages #16060: libupm does not compile with GCC11

neheb, 2021-07-07:
Maintainer: @nxhack 
Environment: GCC11

```
FAILED: src/nrf8001/CMakeFiles/nrf8001.dir/aci_setup.cpp.o
/home/mangix/devstuff/openwrt/staging_dir/host/bin/ccache /home/mangix/devstuff/openwrt/staging_dir/toolchain-mips64el_mips64r2_64_gcc-11.1.0_musl/bin/mips64el-openwrt-linux-musl-g++ -Dnrf8001_EXPORTS -I../include -I../src/nrf8001 -Isrc/nrf8001/.. -Os -pipe -mno-branch-likely -mips64r2 -mtune=mips64r2 -mabi=64 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/home/mangix/devstuff/openwrt/build_dir/target-mips64el_mips64r2_64_musl/upm-2.0.0=upm-2.0.0 -Wformat -Werror=format-security -DPIC -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro  -Wall -Wno-strict-aliasing -Wno-deprecated-declarations -Wno-misleading-indentation -Werror -Wnon-virtual-dtor -Woverloaded-virtual -Wsign-compare -Wreorder -DNDEBUG -fPIC -std=gnu++11 -MD -MT src/nrf8001/CMakeFiles/nrf8001.dir/aci_setup.cpp.o -MF src/nrf8001/CMakeFiles/nrf8001.dir/aci_setup.cpp.o.d -o src/nrf8001/CMakeFiles/nrf8001.dir/aci_setup.cpp.o -c ../src/nrf8001/aci_setup.cpp
../src/nrf8001/aci_setup.cpp: In function 'uint8_t do_aci_setup(aci_state_t*)':
../src/nrf8001/aci_setup.cpp:144:64: error: array subscript 'aci_evt_t[0]' is partly outside array bounds of 'hal_aci_data_t [1]' [-Werror=array-bounds]
  144 |       cmd_status = (aci_status_code_t) aci_evt->params.cmd_rsp.cmd_status;
      |                                        ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
../src/nrf8001/aci_setup.cpp:42:23: note: while referencing 'msg_to_send'
   42 | extern hal_aci_data_t msg_to_send;
      |                       ^~~~~~~~~~~
cc1plus: all warnings being treated as errors
```

nxhack, 2021-07-07:
I cannot prepare an environment to test with gcc-11. I'll do it later.

neheb, 2021-07-07:
My test was with https://github.com/openwrt/openwrt/pull/4343

neheb, 2021-07-15:
This was sort of fixed. At least it compiles now.

closed 2021-07-15 by neheb