openwrt/packages #16080: Gerbera build fail

rektide, 2021-07-08:
# Issue template (remove lines from top till here)

Maintainer: @neheb 
Environment: Qualcomm IPQ806X, Netgear R7800 X4S, Trunk
Compiler: GCC 10

Description: 

```
make[3]: Entering directory '/usr/local/src/openwrt/feeds/packages/multimedia/gerbera'
mkdir -p /usr/local/src/openwrt/dl
SHELL= flock /usr/local/src/openwrt/tmp/.gerbera-1.8.2.tar.gz.flock -c '        /usr/local/src/openwrt/scripts/download.pl "/usr/local/src/openwrt/dl" "gerbera-1.8.2.tar.gz" "b1e5591515f05a46de052ef4d95a2bb338
7e96f565a5ce7abb6a77dbe581f09a" "" "https://codeload.github.com/gerbera/gerbera/tar.gz/v1.8.2?"    '
rm -f /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gerbera-1.8.2/.built
touch /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gerbera-1.8.2/.built_check
MAKEFLAGS="" /usr/local/src/openwrt/staging_dir/host/bin/ninja  -j1 -C /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gerbera-1.8.2
ninja: Entering directory `/usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gerbera-1.8.2'
[16/16] Linking CXX executable gerbera
FAILED: gerbera
: && /usr/local/src/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-10.3.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-
set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gerbera-1.8.2=gerbera-1.8.2 -Wformat -Werror=format-securit
y -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/usr/local/src/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libiconv-stub/include -I/usr/local/src/openwrt/staging_di
r/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libintl-stub/include -ffunction-sections -fdata-sections -flto -DNDEBUG -Wl,--no-as-needed -L/usr/local/src/openwrt/staging_dir/toolchain-arm_cortex-a15+neo
n-vfpv4_gcc-10.3.0_musl_eabi/usr/lib -L/usr/local/src/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-10.3.0_musl_eabi/lib -znow -zrelro -L/usr/local/src/openwrt/staging_dir/target-arm_cortex-a15+n
eon-vfpv4_musl_eabi/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/usr/local/src/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libiconv-stub/lib -L/usr/local/src/openwrt/staging_dir/target
-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libintl-stub/lib -Wl,-rpath-link=/usr/local/src/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libintl-stub/lib -Wl,--gc-sections CMakeFiles
/gerbera.dir/src/main.cc.o -o gerbera  liblibgerbera.a  -latomic  /usr/local/src/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libiconv-stub/lib/libiconv.a  /usr/local/src/openwrt/stag
ing_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libuuid.so  -lnpupnp  /usr/local/src/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libspdlog.a  /usr/local/src/openwrt/stagin
g_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libfmt.so.8.0.1  -Wl,--as-needed  /usr/local/src/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libpugixml.a  /usr/local/src/ope
nwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libsqlite3.so  -lcurl  -L/usr/local/src/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib -ltag -lz  /usr/local/src/open
wrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libmagic.so  /usr/local/src/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libavformat.so  /usr/local/src/openwrt/stag
ing_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libavutil.so  /usr/local/src/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libexif.so  /usr/local/src/openwrt/staging_dir/tar
get-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libebml.a  /usr/local/src/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib/libmatroska.a  /usr/local/src/openwrt/staging_dir/target-arm_cor
tex-a15+neon-vfpv4_musl_eabi/usr/lib/libz.so && /usr/local/src/openwrt/staging_dir/host/bin/cmake -E __run_co_compile --lwyu=gerbera && :
/usr/local/src/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-10.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/10.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: /usr/local/src/openwrt
/tmp/gerbera.xbfkoY.ltrans6.ltrans.o: in function `ContentManager::_rescanDirectory(std::shared_ptr<AutoscanDirectory>&, int, std::shared_ptr<GenericTask> const&)':
<artificial>:(.text._ZN14ContentManager16_rescanDirectoryERSt10shared_ptrI17AutoscanDirectoryEiRKS0_I11GenericTaskE+0x684): undefined reference to `std::__shared_ptr<std::filesystem::_Dir, (__gnu_cxx::_Lock_po
licy)2>::swap(std::__shared_ptr<std::filesystem::_Dir, (__gnu_cxx::_Lock_policy)2>&)'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:88: /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gerbera-1.8.2/.built] Error 1
make[3]: Leaving directory '/usr/local/src/openwrt/feeds/packages/multimedia/gerbera'
```

neheb, 2021-07-08:
This I know about. It's a problem with both GCC10 and 11. I have no solution for it.

neheb, 2021-07-18:
sort of found the issue. The problem is that libmatroska.a is built without being linked to libebml. And I can't figure out why...