openwrt/packages #16081: Freetype build failure

rektide, 2021-07-08:
Maintainer: @neheb @val-kulkov 
Environment: Qualcomm IPQ806X, Netgear R7800 X4S, Trunk
Compiler: GCC 10

Description: 

```
make[3]: Entering directory '/usr/local/src/openwrt/feeds/packages/libs/freetype'
mkdir -p /usr/local/src/openwrt/dl
SHELL= flock /usr/local/src/openwrt/tmp/.freetype-2.10.4.tar.xz.flock -c '      /usr/local/src/openwrt/scripts/download.pl "/usr/local/src/openwrt/dl" "freetype-2.10.4.tar.xz" "86a854d8905b19698bbc8f23b860bc10
4246ce4854dcea8e3b0fb21284f75784" "" "@SF/freetype"    '
rm -f /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/freetype-2.10.4/.built
touch /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/freetype-2.10.4/.built_check
C /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/freetype-2.10.4/openwrt-build
bash: C: command not found
make[3]: [Makefile:68: /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/freetype-2.10.4/.built] Error 127 (ignored)
DESTDIR="/usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/freetype-2.10.4/ipkg-install"  -C /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/freetype-2.10.
4/openwrt-build install
bash: -C: command not found
make[3]: *** [Makefile:70: /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/freetype-2.10.4/.built] Error 127
make[3]: Leaving directory '/usr/local/src/openwrt/feeds/packages/libs/freetype'
```

neheb, 2021-07-08:
Your base is outdated. What's failing is this: https://github.com/openwrt/packages/blob/master/devel/meson/meson.mk#L134

NINJA is defined in https://github.com/openwrt/openwrt/blob/master/rules.mk#L345

rektide, 2021-07-09:
Apologies. Could entirely still be a user error here. I did however delete & re-download my openwrt folder two days ago, so I don't think my base is out of date.

I'm a neophyte with Makefiles, but is it weird that the `Build/Compile/Meson` command isn't just missing the ninja commands, it's missing the `-` in `-C` and doesn't have a `$(1)` in the output?

neheb, 2021-07-09:
What's weird is that it's broken for you. Check your rules.mk file.

rektide, 2021-07-19:
Host computer (Deibna) hasn't been updated in a while. Trying to update some packages. Ok. Updated make 4.2 -> 4.3, meson 0.45 -> 0.57, ninja 1.8 -> 1.10, cmake 3.15->3.18. I know at least 1/2 these aren't relevant. But Freetype builds now. Thank you greatly for your time & investigation.

closed 2021-07-19 by rektide