openwrt/packages #16082: LuaJit: asm/errno.h: No such file or directory

rektide, 2021-07-09:
Maintainer: @neheb 
Environment: Qualcomm IPQ806X, Netgear R7800 X4S, master
Compiler: GCC 10

Hey. Thanks ever so much neheb for the fast replies. Nothing crucial to me to get done, but hoping these are on the radar & available for people/other users to track. Thanks again.

Description:

```
make[3]: Entering directory '/usr/local/src/openwrt/feeds/packages/lang/luajit'
mkdir -p /usr/local/src/openwrt/dl
SHELL= flock /usr/local/src/openwrt/tmp/.LuaJIT-2.1.0-beta3.tar.gz.flock -c '   /usr/local/src/openwrt/scripts/download.pl "/usr/local/src/openwrt/dl" "LuaJIT-2.1.0-beta3.tar.gz" "1ad2e34b111c802f9d0cdf019e986
909123237a28c746b21295b63c9e785d9c3" "" "https://luajit.org/download"    '
touch /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3/.prepared_2223bfa89b907c4f7cfc8c2ab86125e0_6664517399ebbbc92a37c5bb081b5c53_check
. /usr/local/src/openwrt/include/shell.sh; gzip -dc /usr/local/src/openwrt/dl/LuaJIT-2.1.0-beta3.tar.gz | tar -C /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3/.
. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3

Applying ./patches/010-lua-path.patch using plaintext:
patching file src/luaconf.h

Applying ./patches/020-clang.patch using plaintext:
patching file src/lj_arch.h
touch /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3/.prepared_2223bfa89b907c4f7cfc8c2ab86125e0_6664517399ebbbc92a37c5bb081b5c53
rm -f /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3/.configured_*
rm -f /usr/local/src/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.luajit_installed
(cd /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3/./; if [ -x ./configure ]; then find /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_ea
bi/LuaJIT-2.1.0-beta3/ -name config.guess | xargs -r chmod u+w; find /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3/ -name config.guess | xargs -r -n1 cp --remov
e-destination /usr/local/src/openwrt/scripts/config.guess; find /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3/ -name config.sub | xargs -r chmod u+w; find /usr/
local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3/ -name config.sub | xargs -r -n1 cp --remove-destination /usr/local/src/openwrt/scripts/config.sub; AR="arm-openwrt-lin
ux-muslgnueabi-gcc-ar" AS="arm-openwrt-linux-muslgnueabi-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-m
ap=/usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3=LuaJIT-2.1.0-beta3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,rel
ro" LD=arm-openwrt-linux-muslgnueabi-ld NM="arm-openwrt-linux-muslgnueabi-gcc-nm" CC="arm-openwrt-linux-muslgnueabi-gcc" GCC="arm-openwrt-linux-muslgnueabi-gcc" CXX="arm-openwrt-linux-muslgnueabi-g++" RANLIB="
arm-openwrt-linux-muslgnueabi-gcc-ranlib" STRIP=arm-openwrt-linux-muslgnueabi-strip OBJCOPY=arm-openwrt-linux-muslgnueabi-objcopy OBJDUMP=arm-openwrt-linux-muslgnueabi-objdump SIZE=arm-openwrt-linux-muslgnueab
i-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/usr/local/src/openwrt/build_dir/target-arm_co
rtex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3=LuaJIT-2.1.0-beta3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-p
lt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3=
LuaJIT-2.1.0-beta3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/usr/local/src/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-10.3.0
_musl_eabi/usr/include -I/usr/local/src/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-10.3.0_musl_eabi/include/fortify -I/usr/local/src/openwrt/staging_dir/toolchain-arm_cortex-a15+neon[120/1913]
-10.3.0_musl_eabi/include " LDFLAGS="-L/usr/local/src/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-10.3.0_musl_eabi/usr/lib -L/usr/local/src/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfp
v4_gcc-10.3.0_musl_eabi/lib -znow -zrelro "   ./configure --target=arm-openwrt-linux --host=arm-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/us
r --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3/.built
touch /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3/.built_check
make -j1 -C /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3 HOST_CC="gcc -O2 -I/usr/local/src/openwrt/staging_dir/host/include -I/usr/local/src/openwrt/staging_di
r/hostpkg/include -I/usr/local/src/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/include -m32" CROSS="arm-openwrt-linux-muslgnueabi-" DPREFIX=/usr/local/src/openwrt/build_dir/target-arm_c
ortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3/ipkg-install/usr PREFIX=/usr TARGET_SYS=Linux TARGET_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=
unused-result -mfloat-abi=hard -fmacro-prefix-map=/usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3=LuaJIT-2.1.0-beta3 -Wformat -Werror=format-security -fstack-prot
ector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" BUILDMODE=dynamic
make[4]: Entering directory '/usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3'
==== Building LuaJIT 2.1.0-beta3 ====
make -C src
make[5]: Entering directory '/usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3/src'
HOSTCC    host/minilua.o
In file included from /usr/include/bits/errno.h:26,
                 from /usr/include/errno.h:28,
                 from host/minilua.c:40:
/usr/include/linux/errno.h:1:10: fatal error: asm/errno.h: No such file or directory
    1 | #include <asm/errno.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[5]: *** [Makefile:692: host/minilua.o] Error 1
make[5]: Leaving directory '/usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3/src'
make[4]: *** [Makefile:113: default] Error 2
make[4]: Leaving directory '/usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3'
make[3]: *** [Makefile:92: /usr/local/src/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/LuaJIT-2.1.0-beta3/.built] Error 2
make[3]: Leaving directory '/usr/local/src/openwrt/feeds/packages/lang/luajit'
time: package/feeds/packages/luajit/compile#0.37#0.09#0.41
    ERROR: package/feeds/packages/luajit failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/luajit/compile] Error 1
```

neheb, 2021-07-15:
Sounds like your kernel-headers are messed up. What OS are you building on?

rektide, 2021-07-19:
Apologies, I don't always update this particular build system enough. Debian. I upgraded  from linux-libc-dev 4.14.7 to 5.10.46. This resolved luajit building issue.

closed 2021-07-19 by rektide