openwrt/packages #16083: gnutls: 3.7.2 doesn't build

stintel, 2021-07-09:
Maintainer: @nmav  /cc @neheb 
Environment: OpenWrt master r17096-b0483b19f9 on realtek

Description:

3.7.2 bump causes compile error, reverting 59a39d4c6fbe9c0179b16a714bf1a0c3afcbef5e solves the problem.

```
make[4]: Entering directory '/home/build/openwrt/build_dir/target-mips_4kec_musl/gnutls-3.7.2'
make  all-recursive
make[5]: Entering directory '/home/build/openwrt/build_dir/target-mips_4kec_musl/gnutls-3.7.2'
Making all in gl
make[6]: Entering directory '/home/build/openwrt/build_dir/target-mips_4kec_musl/gnutls-3.7.2/gl'
make  all-recursive
make[7]: Entering directory '/home/build/openwrt/build_dir/target-mips_4kec_musl/gnutls-3.7.2/gl'
make[8]: Entering directory '/home/build/openwrt/build_dir/target-mips_4kec_musl/gnutls-3.7.2/gl'
  CC       bitrotate.lo
  CC       cloexec.lo
  CC       hash.lo
  CC       malloca.lo
  CC       read-file.lo
  CC       stat-time.lo
  CC       sys_socket.lo
  CC       unistd.lo
  CC       xsize.lo
  CC       asnprintf.lo
  CC       fcntl.lo
  CC       getdelim.lo
  CC       getline.lo
  CC       malloc.lo
  CC       printf-args.lo
  CC       printf-parse.lo
  CC       realloc.lo
  CC       vasnprintf.lo
In file included from bitrotate.h:23,
                 from bitrotate.c:3:
./stdint.h:89:5: error: #if with no expression
   89 | #if
      |     ^
In file included from /home/build/openwrt/staging_dir/toolchain-mips_4kec_gcc-10.3.0_musl/include/endian.h:23,
                 from /home/build/openwrt/staging_dir/toolchain-mips_4kec_gcc-10.3.0_musl/include/sys/types.h:70,
                 from ./sys/types.h:39,
                 from sys/socket.h:51,
                 from sys_socket.c:3:
./stdint.h:89:5: error: #if with no expression
   89 | #if
      |     ^
In file included from /home/build/openwrt/staging_dir/toolchain-mips_4kec_gcc-10.3.0_musl/include/endian.h:23,
                 from /home/build/openwrt/staging_dir/toolchain-mips_4kec_gcc-10.3.0_musl/include/sys/types.h:70,
                 from ./sys/types.h:39,
                 from ./sys/stat.h:41,
                 from stat-time.h:27,
                 from stat-time.c:3:
./stdint.h:89:5: error: #if with no expression
   89 | #if
      |     ^
make[8]: *** [Makefile:2424: bitrotate.lo] Error 1
make[8]: *** Waiting for unfinished jobs....
In file included from xsize.h:27,
                 from xsize.c:3:
./stdint.h:89:5: error: #if with no expression
   89 | #if
      |     ^
In file included from /home/build/openwrt/staging_dir/toolchain-mips_4kec_gcc-10.3.0_musl/include/endian.h:23,
                 from /home/build/openwrt/staging_dir/toolchain-mips_4kec_gcc-10.3.0_musl/include/sys/types.h:70,
                 from ./sys/types.h:39,
                 from ./stdio.h:58,
                 from read-file.h:25,
                 from read-file.c:20:
./stdint.h:89:5: error: #if with no expression
[... 109 lines cut ...]
make[8]: *** [Makefile:2424: hash.lo] Error 1
make[8]: *** [Makefile:2424: printf-parse.lo] Error 1
make[8]: *** [Makefile:2424: cloexec.lo] Error 1
make[8]: *** [Makefile:2424: unistd.lo] Error 1
In file included from printf-args.h:41,
                 from printf-parse.h:29,
                 from vasnprintf.c:90:
./wchar.h:621:6: error: #if with no expression
  621 | # if
      |      ^
make[8]: *** [Makefile:2424: printf-args.lo] Error 1
make[8]: *** [Makefile:2424: fcntl.lo] Error 1
make[8]: *** [Makefile:2424: vasnprintf.lo] Error 1
make[8]: Leaving directory '/home/build/openwrt/build_dir/target-mips_4kec_musl/gnutls-3.7.2/gl'
make[7]: *** [Makefile:2445: all-recursive] Error 1
make[7]: Leaving directory '/home/build/openwrt/build_dir/target-mips_4kec_musl/gnutls-3.7.2/gl'
make[6]: *** [Makefile:2277: all] Error 2
make[6]: Leaving directory '/home/build/openwrt/build_dir/target-mips_4kec_musl/gnutls-3.7.2/gl'
make[5]: *** [Makefile:2225: all-recursive] Error 1
make[5]: Leaving directory '/home/build/openwrt/build_dir/target-mips_4kec_musl/gnutls-3.7.2'
make[4]: *** [Makefile:2150: all] Error 2
make[4]: Leaving directory '/home/build/openwrt/build_dir/target-mips_4kec_musl/gnutls-3.7.2'
make[3]: *** [Makefile:265: /home/stijn/Development/OpenWrt/openwrt/build_dir/target-mips_4kec_musl/gnutls-3.7.2/.built] Error 2
time: package/feeds/packages/gnutls/compile#0.95#0.53#0.52
```

stintel, 2021-07-20:
Please look into a fix, if not I will revert the bump soonish.

neheb, 2021-07-20:
IIRC, I was getting a similar issue on Arch Linux. Can't reproduce on Fedora. I'll have to spin up a VM.

neheb, 2021-07-20:
Please test: https://github.com/neheb/packages/commit/9eeed5bd70ee50a73d8d30e164c0c52db81f4f98

stintel, 2021-07-21:
> Please test: [neheb@9eeed5b](https://github.com/neheb/packages/commit/9eeed5bd70ee50a73d8d30e164c0c52db81f4f98)

That seems to fix the problem for me, thanks!

stintel, 2021-07-21:
> > Please test: [neheb@9eeed5b](https://github.com/neheb/packages/commit/9eeed5bd70ee50a73d8d30e164c0c52db81f4f98)
> 
> That seems to fix the problem for me, thanks!

Scratch that. I must have forgotten make/package/gnutls clean.

commit 3d4e6b05 (2021-08-05): Revert "gnutls: update to 3.7.2"

Fixes #16083

This reverts commit 59a39d4c6fbe9c0179b16a714bf1a0c3afcbef5e.
closed 2021-08-05 by stintel