openwrt/packages #16085: saned xinetd: regression on openwrt 21

blairfancy, 2021-07-10:
# Issue template (remove lines from top till here)

Maintainer: @luizluca ; @SvenRoederer
Environment: openwrt-21.02.0-rc3-mvebu-cortexa9

Description:
installed saned and xinetd package modified sane.conf and xinetd config with same parameters thant openwrt 19.
Scanner work properly with

```
scanimage -T -o toto.jpg
```

But not able to scan with a remote Host (192.168.25.10)

xinetd
```
config service 'saned_service'
        option socket_type 'stream'
        list only_from '192.168.25.10'
        option port '6566'
        option server '/usr/sbin/saned'
        option type 'UNLISTED'
        option user 'saned'
        option protocol 'tcp'
        option instances 'UNLIMITED'
        option wait 'no'
        option disable 'no'
        option server_args '--debug=6'

```
saned.conf
```
192.168.25.0/24

```

log
```
Sat Jul 10 17:08:24 2021 daemon.debug xinetd[25733]: Reading included configuration file: /etc/xinetd.d/sane-port [file=/var/run/xinetd.conf] [line=9]
Sat Jul 10 17:08:24 2021 daemon.debug xinetd[25733]: removing sane-port
Sat Jul 10 17:08:24 2021 daemon.notice xinetd[25733]: xinetd Version 2.3.15 started with loadavg options compiled in.
Sat Jul 10 17:08:24 2021 daemon.notice xinetd[25733]: Started working: 1 available service
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: read_config: searching for config file
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: read_config: done reading config
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: saned (AF-indep+IPv6) from sane-backends 1.0.32 starting up
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: handle_connection: processing client connection
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: access by remote host: 192.168.25.10
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: remote host is not IN_LOOPBACK nor IN6_LOOPBACK
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: local hostname: OpenWrt
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: local hostname(s) (from DNS): OpenWrt.lan
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: local hostname(s) (from DNS): OpenWrt.lan
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: local hostname(s) (from DNS): OpenWrt.lan
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: local hostname(s) (from DNS): OpenWrt.lan
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: remote host doesn't have same addr as local
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: opening config file: /etc/hosts.equiv
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: can't open config file: /etc/hosts.equiv (No such file or directory)
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: opening config file: saned.conf
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `# saned.conf'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `# Configuration for the saned daemon'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `## Daemon options'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `# Port range for the data connection. Choose a range inside [1024 - 65535].'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `# Avoid specifying too large a range, for performance reasons.'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `#'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `# ONLY use this if your saned server is sitting behind a firewall. If your'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `# firewall is a Linux machine, we strongly recommend using the'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `# Netfilter nf_conntrack_sane connection tracking module instead.'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `#'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `# data_portrange = 10000 - 10100'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `## Access list'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `# A list of host names, IP addresses or IP subnets (CIDR notation) that'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `# are permitted to use local SANE devices. IPv6 addresses must be enclosed'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `# in brackets, and should always be specified in their compressed form.'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `#'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `# The hostname matching is not case-sensitive.'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `#scan-client.somedomain.firm'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `#192.168.0.1'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: config file line: `192.168.25.0/24'
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: subnet with base IP = 192.168.25.0, CIDR netmask = 24
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: check_host: access granted from IP address 192.168.25.10 (in subnet 192.168.25.0/24)
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: init: access granted
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: init: access granted to blairfancy@192.168.25.10
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: process_request: waiting for request
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: process_request: got request 1
Sat Jul 10 17:08:51 2021 daemon.debug saned[25843]: process_request: waiting for request
Sat Jul 10 17:08:52 2021 daemon.debug saned[25843]: process_request: got request 10
Sat Jul 10 17:08:52 2021 daemon.debug saned[25843]: saned exiting
```

luizluca, 2021-07-12:
I'm preparing a test env but I lost my old printer #16111 

It looks like the network side is OK. You can connect, be authenticated (src ip) and do some requests. My guess is that the printer driver is not working. You can list existing printers with '`scanimage -L`' in the client (or in router if you install sane-frontends). It will probably not list your scanner.

You could also install sane-test to test with a fake printer (you need to uncomment "test" in /etc/sane.d/dll.conf). Currently, the sane-test is all I have to work on.

luizluca, 2021-07-12:
At least for sane-test, everything is still working.

```
$ scanimage -L
device `net:192.168.1.1:test:0' is a Noname frontend-tester virtual device
device `net:192.168.1.1:test:1' is a Noname frontend-tester virtual device
$ scanimage -T 
Output format is not set, using pnm as a default.
scanimage: scanning image of size 157x196 pixels at 8 bits/pixel
scanimage: acquiring gray frame, 8 bits/sample
scanimage: reading one scanline, 157 bytes...   PASS
scanimage: reading one byte...          PASS
scanimage: stepped read, 2 bytes...     PASS
scanimage: stepped read, 4 bytes...     PASS
scanimage: stepped read, 8 bytes...     PASS
scanimage: stepped read, 16 bytes...    PASS
scanimage: stepped read, 32 bytes...    PASS
scanimage: stepped read, 64 bytes...    PASS
scanimage: stepped read, 128 bytes...   PASS
scanimage: stepped read, 256 bytes...   PASS
scanimage: stepped read, 255 bytes...   PASS
scanimage: stepped read, 127 bytes...   PASS
scanimage: stepped read, 63 bytes...    PASS
scanimage: stepped read, 31 bytes...    PASS
scanimage: stepped read, 15 bytes...    PASS
scanimage: stepped read, 7 bytes...     PASS
scanimage: stepped read, 3 bytes...     PASS
$ scanimage -o toto.jpg
$ file toto.jpg 
toto.jpg: JPEG image data, JFIF standard 1.01, resolution (DPI), density 50x50, segment length 16, baseline, precision 8, 157x196, components 1
```

It should be something with your driver. Are you using hplip?

blairfancy, 2021-07-13:
On server side scanimage -L return : 
`device 'genesys:libusb:002:003' is a Canon LiDE 35/40/50 flatbed scanner`

on computer scanimage -L

`No scanners were identified. If you were expecting something different,
check that the scanner is plugged in, turned on and detected by the
sane-find-scanner tool (if appropriate). Please read the documentation
which came with this software (README, FAQ, manpages).
`

/etc/sane.d/net.conf
`192.168.25.1`

Witch packages should be installed?

blairfancy, 2021-07-13:
I confirm that test devices are seen by my computer
```
device 'net:192.168.25.1:test:0' is a Noname frontend-tester virtual device
device 'net:192.168.25.1:test:1' is a Noname frontend-tester virtual device
```

luizluca, 2021-07-14:
That's very useful info!

test scanner works, so your client/server setup is OK. It is something between the daemon and the scanner.
scanimage works, which means you installed the correct driver. It does not use daemon.

The issue might be with the user saned accessing directly the usb device. This is granted using a hotplug.d script that check if the device usb id matches a file in /usr/share/sane/.

Now the interesting part... the file build by buildbot has:

```
./
./etc/
./etc/sane.d/
./etc/sane.d/genesys.conf
./usr/
./usr/lib/
./usr/lib/sane/
./usr/lib/sane/libsane-genesys.so.1
./usr/lib/sane/libsane-genesys.so.1.0.32
```
While the one I build locally has:

```
./
./etc/
./etc/sane.d/
./etc/sane.d/genesys.conf
./usr/
./usr/lib/
./usr/lib/sane/
./usr/lib/sane/libsane-genesys.so.1
./usr/lib/sane/libsane-genesys.so.1.0.32
./usr/share/
./usr/share/sane/
./usr/share/sane/03f0-genesys.usbid
./usr/share/sane/0461-genesys.usbid
./usr/share/sane/04a7-genesys.usbid
./usr/share/sane/04a9-genesys.usbid
./usr/share/sane/04da-genesys.usbid
./usr/share/sane/04f9-genesys.usbid
./usr/share/sane/07b3-genesys.usbid
./usr/share/sane/0a17-genesys.usbid
./usr/share/sane/0a82-genesys.usbid
./usr/share/sane/1083-genesys.usbid
./usr/share/sane/1dcc-genesys.usbid
```

Noticed the usbid files missing? They are the ones you'll need. I'll check build logs to try to identify why they are missing...

Meanwhile, you can create the file manually. If your device usbid is something like 04a9:xxxx, you can create a file /usr/share/sane/04a9-genesys.usbid with xxxx inside. For example, the expected file to be in the package would be:

```
/usr/share/sane/04a9-genesys.usbid:
1900
1904
1905
1906
1907
1909
190a
190e
190f
2213
2214
221c
221e
2228
2229
```

Among other vendor ids. The file list from the package I built locally shows them all. The suffix does not really matter as I only use that info for selecting usbid files to each driver package.

Another alternative is to run as root, changing the user in xinetd config.

closed 2021-07-18 by luizluca

luizluca, 2021-07-18:
@blairfancy, the fix was merged. The fix will be in the 1.0.32-2 package version. It might reach repos soon.

Thanks @micmac1 for the fix.

PR #16142 (2021-07-18): sane-backends: fix usbid generation
Maintainer: @luizluca 
Compile tested: master ath79
Run tested: N/A

Description:
Hi Luiz,

I could repro #16085 on a Debian box. I fixed it, the usbid files are now generated. The build bots seem to use Debian or a derivative as well, so same issue there. Also added a commit with a few "invocation" fixes on top. I can pop this commit if you don't want it.

Regards,
Seb

Closes #16085
commit 041e2877 (2021-07-18): sane-backends: fix usbid file generation

On some build systems (build bots, Debian Buster for example) the
current mechanism in the Build/Install define doesn't run. Replace it
with shell fu that works.

Issue was reported, see [1].

[1] https://github.com/openwrt/packages/issues/16085

Fixes #16085

(cherry picked from commit 977109e28613820ff451908289ff6e4cc671ec32)