openwrt/packages #16111: sane-backends,hplip,libcups: looking for maintainer

luizluca, 2021-07-12:
I recently lost my printer used for sane-backends and hplip tests. The new one uses proprietary HP driver not compiled for anything but arm/x86.

https://github.com/openwrt/packages/issues/14830
https://forum.openwrt.org/t/sane-issue-on-hp-laserjet-m1132/57698
https://developers.hp.com/printos/forum/proprietary-drivers-hplip-plugins-more-archs?language=ja

Even if HP provided the driver for mips and others, I'm not sure if it will work with OpenWrt nor if the router HW could withstand the driver needs.

I can continue maintaining the package but I'll only be able to test with fake printers (sane-test). However, most issues are with USB/hplip integration. It would be better to someone with real use to take the task.
As my new device is a network printer, I'm not going to use sane,hplip in OpenWrt anymore.

libcups is just a stripped cups used as hplip build requirements (until someone maintains the full cups for OpenWrt).

I'll ping previous committers @neheb @syssi

walterav1984, 2022-04-08:
Since [OpenPrinting CUPS](https://github.com/OpenPrinting/cups/releases/tag/v2.4b1) comes with Airprint, IPP-Everywhere and Mopria support builtin VS the former Apple based CUPS, should the package libcups if rebased to OpenPrinting be considered for re-including non-stripped by default builds in 22.03 release?

luizluca, 2022-04-09:
Cups left repo because no one wanted to maintain it. Simple as that. Sane-backends and hplip might face the same fate if no one take care of them. I might keep updating them but eventually I will break them as I cannot test them anymore.

If someone submit a good full cups or openprinting package, there is no reason for not accepting it. 

Libcups mentioned here is just used for runtime dependency. It does not even have an installable package.

walterav1984, 2022-04-09:
> If someone submit a good full cups or openprinting package, there is no reason for not accepting it.

So first step for me would be try to replace and build 'op-cups' the same as 'cups' with only hplip and see if that compiles than step by step enabling more modules? If it builds and works on 2 or 3 architectures ask you to accept changes?