openwrt/packages #16129: mwan3: iface 15 exit sooner

wulfy23, 2021-07-15:
Maintainer: @feckert @aaronjg 
Environment: na
Description: is it possible to exist a little sooner prior to sourcing all the functions? carried this local fix for a while...

```
if ! /etc/init.d/mwan3 running && ! /etc/init.d/mwan3 enabled; then
    exit 0
fi

. /lib/functions.sh
...
init ...
```

feckert, 2021-08-12:
You could check if the tracker for the interface has been started.
However, this only works if a tracker is configured and is also enabled.
`ubus call service list "{'name':'mwan3', 'verbose': true}"`

The way you do it doesn't work with this [line](https://github.com/openwrt/packages/blob/master/net/mwan3/files/etc/hotplug.d/iface/15-mwan3#L26) I think.

What do you need that for?

wulfy23, 2021-08-12:
Thanks for the tip...

Just minimising code loading/parsing when disabled (overall system minimalism)

Afaik, there are no negative effects letting the following events to proceed... Seemed it could be a little lighter is all...

feckert, 2023-05-03:
Ok let's close this. From my point of view, this is done or not relevant and not an issue.

closed 2023-05-03 by feckert