openwrt/packages #16143: ddns-scripts: preinst exec format error

peci1, 2021-07-17:
Maintainer: @feckert 
Environment: Turris Omnia, arch mvebu, Turris OS 5.2.3 based on 19.07

After updating to the newest version of the OS, I got this error:

```
ddns-scripts_no-ip_com/preinst: [31;1mDIE[0m:Failed to exec /usr/lib/opkg/info//ddns-scripts_no-ip_com.preinst: Exec format error

ddns-scripts/preinst: [31;1mDIE[0m:Failed to exec /usr/lib/opkg/info//ddns-scripts.preinst: Exec format error
```

feckert, 2021-09-03:
Please open an issue on Turris

peci1, 2021-09-03:
Turris developers sent me here. They say they do not adjust this package in any way and they just use the vanilla OpenWRT version. Thus I think the bug report belongs here.

feckert, 2021-09-03:
You'll have to give me more information on how I can reproduce this or when this error occurs.

- On entering a ddns command?
- Output of the ddns service during operation into the log?

Check you could do for me to help you

- Have you ever looked at the script
- Try running a [shell check](https://www.shellcheck.net/) on the script
- Have you compared the script if htis is equal to that in openwrt-21.02 branch in this feed

peci1, 2021-09-03:
This happens during automated upgrade of the system (which is a Turris-custom script using opkg internally).

I'll try to have a look at this, but my time is very limited in September, so I'll probably respond later.

peci1, 2021-11-16:
Another update and now I managed to save the update script: [ddns-scripts_no-ip_com.zip](https://github.com/openwrt/packages/files/7545834/ddns-scripts_no-ip_com.zip). The error is obvious - all lines in the bash script (including shebang) start with a tab character. I don't know where does this script come from, but I believe that removing the tabs would allow executing it.

peci1, 2021-11-21:
It seems to me that the wrong indentation is caused by the Makefile on 19.07: https://github.com/openwrt/packages/blob/openwrt-19.07/net/ddns-scripts/Makefile . You can see a lot of preinst and other sections starting with `\t#!/bin/sh`.  Master branch has somehow gotten rid of these preinst scripts, so this problem is only on 19.07 and 18.06.

feckert, 2021-11-24:
Branch 19.07 will soon be EOL.
If you want this to be fixed, please open a pullerquest with the correct indentation and we can add it to 19.07

peci1, 2021-11-24:
Okay, I can do that. Is the Makefile I pointed to the correct file to fix (I guess it is, but I want to be sure). Should I fix just the no_ip section that troubles me, or should I fix all other sections that seem to have the same problem?

feckert, 2021-11-26:
While we're at it, I would suggest that we fix everything, otherwise the next person will fall over the same problem again!

peci1, 2021-11-26:
Fix proposed in 17209 .

PR #17209 (2021-11-30): ddns-scripts: Fix wrong whitespace in preinst and postinst scripts
Maintainer: unkown?
Compile tested: no
Run tested: no

Description: Fixes #16143 by removing superfluous whitespace at the beginnings of preinst/postinst shell scripts. The scripts are apparently generated as is directly from the makefile, which means the scripts begin with `\t#!/bin/sh` and the interpreter is not happy with that.

Alternatively, it should be enough to unindent the shebang lines only. However, it would seem weird to me.
closed 2022-08-18 by peci1