openwrt/packages #16162: ddns-scripts-digitalocean: Script fails, but updates IPv6 hostnames

giannoug, 2021-07-20:
Maintainer: @feckert (hope I got it right 😨)
Environment: ASUS RT-AC58U / ARMv7 Processor rev 5 (v7l) / OpenWrt 21.02.0-rc3 r16172-2aba3e9784 / LuCI openwrt-21.02 branch git-21.163.64918-ba57ec5

Description:
After setting up and running an IPv6 DDNS service for DigitalOcean with ddns-scripts-digitalocean, the script seems to fail on the first run. The DNS provider has the correct IP assigned, which means that the update is successful, but the ddns-script is reporting a failure in LuCI (ERROR : IP update not accepted by DDNS Provider).

I believe this is happening because IPv6 addresses are compared as strings on [line 50](https://github.com/openwrt/packages/blob/openwrt-21.02/net/ddns-scripts/files/usr/lib/ddns/update_digitalocean_com_v2.sh#L50). This fails due to local IP being compressed and remote IP being expanded.

Sample output (with personal information removed):
```
 155835       : Local IP 'xxxx:xxxx:xxxx:b00::abc' detected on network 'wan_6'
 155835       : Update needed - L: 'xxxx:xxxx:xxxx:0b00:0000:0000:0000:0abc' <> R: '2a00:1450:4017:0803:0000:0000:0000:200e'
 155835       : parsing script '/usr/lib/ddns/update_digitalocean_com_v2.sh'
 155837       : DDNS Provider answered:
Server:

127.0.0.1
Address:
127.0.0.1#53

Name:      hostname.example.com
Address 1: xxx.xxx.xxx.xxx
Address 2: xxxx:xxxx:xxxx:b00::abc
 155837 ERROR : IP update not accepted by DDNS Provider
 155837       : Waiting 600 seconds (Check Interval)
```

feckert, 2021-08-31:
@giannoug 
I have no way of testing this provider. But your suspected error could describe the behaviour.
Perhaps we should also adapt the error handling as in this [pullrequest](https://github.com/openwrt/packages/pull/16430/files)?
What do you get back in the event of an error or success?

giannoug, 2021-08-31:
I believe it should be easier than that: https://docs.digitalocean.com/reference/api/api-reference/#operation/update_domain_record

According to the docs, we can rely solely on the HTTP response status. 200 is success. 401, 404, 429 and 500 can be logged with specific messages to help the user and fail. Maybe we can use curl with `-w "%{http_code}"`?

giannoug, 2021-11-30:
Fixed in #17228 and #17120

closed 2021-11-30 by giannoug