openwrt/packages #16193: need binutils-gold for crowdsec package (GO)

erdoukki, 2021-07-22:
I need binutils-gold for building Crowdsec for OpenWrt !

I already get it on OpenWrt host, for POC ;
https://discourse.crowdsec.net/t/usr-bin-ld-cannot-find-ldl/143
```
root@ultra:~/binutils-2.36# ldd.gold --version
GNU gold (GNU Binutils 2.36) 1.16
Copyright (C) 2021 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
```
now I try to crossbuild a Package (before PR) on an Ubuntu host...

Any one can help me on this, please ?

References : 
https://stackoverflow.com/questions/23589392/trying-to-build-binutils-with-enable-gold
http://llvm.org/docs/GoldPlugin.html

erdoukki, 2021-07-30:
I get the cross build to work with : 
```
[*] Advanced configuration options (for developers)  --->
[*]   Toolchain Options  ---> 
Binutils Version (Binutils 2.34)  --->
(--enable-gold --enable-plugins) Additional binutils configure options
```
which gives me (with grep .config) : 
```
CONFIG_EXTRA_BINUTILS_CONFIG_OPTIONS="--enable-gold --enable-plugins"
```
But I cannot get it in predefined value of my Makefile !
something like : 
https://github.com/openwrt/packages/pull/16244#issuecomment-889383704
```
  PKG_CONFIG_DEPENDS:= \
  	@CONFIG_EXTRA_BINUTILS_CONFIG_OPTIONS="--enable-gold --enable-plugins"
```
How can I tell my Makefile to add this extra binutils config options ?

It's a requirement for ARM build of GO SQLITE3 included in CrowdSec !

erdoukki, 2021-07-31:
May be something more like:
```
PKG_BUILD_DEPENDS:= \
 	BINUTILS:CONFIG_EXTRA_BINUTILS_CONFIG_OPTIONS=\"--enable-gold --enable-plugins\"
```

erdoukki, 2021-07-31:
Stuck on this !
I have tried some differents possible tweaks…
Is it only possible to modify .config from a package Makefile ?
Are the CONFIG sections checks at build only time ?

erdoukki, 2021-07-31:
I need to use this commit https://github.com/openwrt/openwrt/commit/19a6b0fa95f3b38e9019d2151ad77e6affb07947

I want to use and configure from the package Makefile

erdoukki, 2021-07-31:
From https://github.com/mwarning/openwrt-examples

> HOST_CONFIGURE_ARGS: Similar to CONFIGURE_ARGS, but used when build for the host system architecture. This is needed for building the build tools.

so, setting HOST_CONFIGURE_ARGS would be sufiscient ?
```
HOST_CONFIGURE_ARGS += \
	--enable-gold \
	--enable-plugins
```

erdoukki, 2021-07-31:
I suppose that I need to do something like the select for a boolean value, but for a string I do not get success !

erdoukki, 2021-08-01:
@jefferyto
Do you have advice about gold linker support for CGO requirements on ARM64 ?
I will wrote a patch for binutils to support gold-linker and make a PR on OpenWrt ToolChain…

erdoukki, 2021-08-01:
https://github.com/openwrt/openwrt/pull/4412

jefferyto, 2021-08-11:
Many Go programs here are compiled with CGO enabled and for all supported platforms (including arm64). Can you point to where you specifically need the gold linker?

erdoukki, 2021-08-11:
The problem is with the link, only on arm architecture, of go-sqlite3.
https://github.com/mattn/go-sqlite3
This make a run error without CGO
```
root@ultra:~# cd go-sqlite3/
root@ultra:~/go-sqlite3# CGO_ENABLED=1 GOOS=linux GOARCH=arm64 go build
root@ultra:~/go-sqlite3# cscli -c /etc/crowdsec/config.yaml machines add auto
FATA[16-07-2021 07:22:49 PM] unable to create new database client: failed creating schema resources: Binary was compiled with 'CGO_ENABLED=0', go-sqlite3 requires cgo to work. This is a stub
```
and this make a link error with CGO
```
root@ultra:~/crowdsec# CGO_ENABLED=1 GOOS=linux GOARCH=arm64 make release
/root/crowdsec/platform/linux.mk:5: Building for linux
rm -f crowdsec
go build -ldflags "-s -w -X github.com/crowdsecurity/crowdsec/pkg/cwversion.Version="v1.1.1" -X github.com/crowdsecurity/crowdsec/pkg/cwversion.System=linux -X github.com/crowdsecurity/crowdsec/pkg/cwversion.BuildDate=2021-07-17_10:16:25 -X github.com/crowdsecurity/crowdsec/pkg/cwversion.Codename=alphaga -X github.com/crowdsecurity/crowdsec/pkg/cwversion.Tag="1610255cbdbb901435d399f8fd533d7267a24adf" -X github.com/crowdsecurity/crowdsec/pkg/cwversion.GoVersion="1.16.5"" -o crowdsec -v
# github.com/crowdsecurity/crowdsec/cmd/crowdsec
/usr/lib/go-1.16/pkg/tool/linux_arm64/link: running gcc failed: exit status 1
collect2: fatal error: cannot find 'ld'
compilation terminated.

make[1]: *** [Makefile:20: build] Error 2
make: *** [Makefile:91: crowdsec] Error 2
```

erdoukki, 2021-08-11:
I will give a try to better enhancing the usage of goland-pacakage.mk for building crowdsec package...

**but** the problem is not the linker is not found, but that without gold linker, the linking just not works on arms !

I will try to debug (again) this issue.

jefferyto, 2021-08-11:
I found [one case](https://github.com/golang/go/blob/go1.16.7/src/cmd/link/internal/ld/lib.go#L1401) where the Go compiler will switch to the gold linker for arm, when it is building code for dynamic linking. The Go compiler will do dynamic linking if `-buildmode=shared` or `-buildmode=plugin` (or `-linkshared`) is specified on the command line, or if it thinks the program it is compiling will (dynamically) load plugins. And I believe it figures out if the program will load plugins by looking to see if the `plugin` package is used.

In crowdsec I found [one place](https://github.com/crowdsecurity/crowdsec/blob/v1.1.1/pkg/parser/enrich.go#L4) where the `plugin` package is used, but from the code (and code comments) it appears that it is not doing real plugin loading. I believe if you patch that code to remove references to the `plugin` package, then the Go compiler will not switch to the gold linker for arm.

erdoukki, 2021-08-12:
@jefferyto, thanks for your analysis, I am not a Go expert... ;-)
I have removed the 2 references to plugin, and the compilation still works !
Some small tests give the tools look also to still works !
I have pushed the modified sources with a new patch to see if the toolchain will get it compile on GITHUB-CI...
I will give also an issue at CrowdSec GITHUB.

erdoukki, 2021-08-12:
Okay now, except for unsuported platform : 
```
cmd/go: unsupported GOOS/GOARCH pair linux/arc
```
But nothing to do with binutils-gold !

jefferyto, 2021-08-12:
I think the source file packages also need `DEPENDS:=$(GO_ARCH_DEPENDS)`.

erdoukki, 2021-08-12:
> I think the source file packages also need `DEPENDS:=$(GO_ARCH_DEPENDS)`.

yes ! thanks...

> `cmd/go: unsupported GOOS/GOARCH pair linux/arc`

Fixed now...

If you can look at the PR, please, I have new issue with the "standard" build method : 
https://github.com/openwrt/packages/pull/16244#issuecomment-897509362

binutils-gold may not be needed with your tweak of disabling the plugin from https://github.com/openwrt/packages/issues/16193#issuecomment-897126602
I will report to CrowdSec...

closed 2021-08-12 by erdoukki

adonespitogo, 2023-09-02:
I have a similar problem. But in my case this is a custom application that needs to be built with `buildmode=plugin` so I can't just remove the `plugin` import. I'm not experienced in cross-compilation, what could be the possible solution?

jefferyto, 2023-09-04:
From https://github.com/golang/go/issues/22040 it appears that with binutils 2.41 it will be safe to remove the use of the gold linker. I will wait until the main repo selects that version by default (the [current default](https://github.com/openwrt/openwrt/blob/8554a4a7e37998bd214666d2ab7e469ce7c8757e/toolchain/binutils/Config.in#L5) is 2.40) to apply the patch linked on that issue. If you need to have it right now you can try applying that patch to the golang package and ensuring binutils 2.41 is selected for your build.

adonespitogo, 2025-01-07:
@jefferyto do you know if there's any ongoing work to upgrade `binutils` to 2.41 on OpenWrt?