openwrt/packages #16199: /usr/sbin/mwan3rtmon ipv6 is using all of CPUs

zhihuiyuze, 2021-07-22:
Hi,I have upgrade to  openwrt-21.02 branch (git-21.196.62745-ea91a4e).And all of the rules is adding 
`option family 'ipv4'` for diable ipv6.
At first i think is caues by too may ipv6 route tables,So stop bird2 so there is no too may route table but 
`/usr/sbin/mwan3rtmon ipv6` still using 100% cpu.
i check the log,It seemed to be the ipv6 getting from the PPPoe interface.
So i create manually by tracking is the PPPoe interface and using dhcpv6,but it doesn't work.
Even i shouting down all of ipv6 and `service mwan3 stop`and still 100cpu,mwan3rtmon have not been killed.
[image]

```
Fri Jul 23 02:38:06 2021 daemon.notice netifd: Interface 'WAN2_6_4' is now down
Fri Jul 23 02:38:06 2021 daemon.notice netifd: Interface 'WAN2_6_4' is setting up now
Fri Jul 23 02:38:07 2021 daemon.notice netifd: Interface 'wan_6_4' is now down
Fri Jul 23 02:38:07 2021 daemon.notice netifd: Interface 'wan_6_4' is setting up now
Fri Jul 23 02:38:08 2021 daemon.info https-dns-proxy[12574]: [E] 1626979088.675606 https_client.c:205 No response
Fri Jul 23 02:38:09 2021 user.info : info: guest-ping called
Fri Jul 23 02:38:09 2021 daemon.notice netifd: Interface 'WAN2_6_4' is now down
Fri Jul 23 02:38:09 2021 daemon.notice netifd: Interface 'WAN2_6_4' is setting up now
Fri Jul 23 02:38:10 2021 daemon.notice netifd: Interface 'wan_6_4' is now down
Fri Jul 23 02:38:10 2021 daemon.notice netifd: Interface 'wan_6_4' is setting up now

```

```
[10300.718872] r8152 2-1:1.0 eth1: carrier on
[10301.098741] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[10301.103323] r8152 2-1:1.0 eth1: carrier on
[10301.402470] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[10301.406797] r8152 2-1:1.0 eth1: carrier on
[10301.754590] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[10301.758470] r8152 2-1:1.0 eth1: carrier on
[10302.043866] r8152 2-1:1.0 eth1: carrier on
[10302.129044] 8021q: adding VLAN 0 to HW filter on device eth0
[10302.304768] pppoe-wan2: renamed from ppp0
[10302.324675] pppoe-wan: renamed from ppp1
[10344.521084] Set mwan3_connected_v6 is full, maxel

```
```
config rule 'IX_VM_china'
        option src_ip '192.168.7.57,192.168.7.54,192.168.7.55,192.168.7.59,192.168.7.63,192.168.7.67,192.168.7.69,192.168.7.70,192.168.7.71,192.168.7.72,192.168.7.73'
        option proto 'all'
        option ipset 'cn'
        option family 'ipv4'
        option dest_ip '0.0.0.0/0'
        option use_policy 'balanced'
        option sticky '1'

config rule 'IX_VM'
        option family 'ipv4'
        option proto 'all'
        option sticky '0'
        option src_ip '192.168.7.57,192.168.7.54,192.168.7.55,192.168.7.59,192.168.7.63,192.168.7.67,192.168.7.69,192.168.7.70,192.168.7.71,192.168.7.72,192.168.7.73'
        option use_policy 'warp'
        option dest_ip '0.0.0.0/0'

config rule 'vpn_ca'
        option dest_ip '104.218.61.232'
        option sticky '0'
        option use_policy 'balanced'
        option family 'ipv4'
        option proto 'all'

config rule 'vpn_hk'
        option dest_ip '103.152.35.35'
        option sticky '0'
        option use_policy 'balanced'
        option family 'ipv4'
        option proto 'all'

[... 298 lines cut ...]
        option count '1'
        option size '56'
        option max_ttl '60'
        option check_quality '0'
        option timeout '4'
        option interval '10'
        option failure_interval '5'
        option recovery_interval '5'
        option down '5'
        option up '5'
```
```

config interface 'WAN2_6'
        option device '@wan2'
        option proto 'dhcpv6'
        option reqaddress 'try'
        option reqprefix '60'

config interface 'wan_6'
        option proto 'dhcpv6'
        option reqprefix '60'
        option reqaddress 'try'
        option device '@wan'
```

zhihuiyuze, 2021-07-24:
HI,I have tried to diebles bird2 so the time when the system start,there was no more routing table from bird.Uniil now,there are no `/usr/sbin/mwan3rtmon ipv6` is using 100 cpu.
But i have a few qusetions,Is mwan3 loading all of route4 and route6 when it starting?And it seems mwan3 is useing iptables to archieve the purpose of diversion.Why it needs to load routes?Also iptables has higher priority then routing table,so it doesn't needs to load routes.
Could you please help me find the `ipv6` function in `/usr/sbin/mwan3rtmon ipv6` ? because i can't find in the code.

feckert, 2023-09-20:
Sorry for the late reply. The release owrt-21.02 is already EOL. 

> Is mwan3 loading all of route4 and route6 when it starting?

I'm afraid I don't understand exactly what you mean. But when the mwan3 starts it syncs all static routings and the default route for the interface into the dedicated routing table of the interface. After that the mwanrtmon takes over and syncs them on route changes.

> And it seems mwan3 is useing iptables to archieve the purpose of diversion.

Yes iptables marks the packets with a ruleset on which interface routing table the package should ask. Since owrt-23.05 and master nftables is used for this task via iptables-nft.

> Why it needs to load routes?

I do not understand

> Also iptables has higher priority then routing table,so it doesn't needs to load routes.

I do not understand

> Could you please help me find the `ipv6` function in `/usr/sbin/mwan3rtmon ipv6` ? because i can't find in the code.

There is a service `/usr/sbin/mwan3rtmon ipv6` which is started via `/etc/init.d/mwan3`. 

I'm going to close the issue now, if you still want to know something then feel free to reply to the closed ticket.

closed 2023-09-20 by feckert