openwrt/packages #16216: banip: Isn't having a reference to table 'banIP' in table 'input_lan_rule' obsolete?

manuel-kraus-19, 2021-07-24:
Maintainer: @dibdot
Environment: mips, TP-Link Archer C7 v4, 19.07.7

Description:

I'm currently trying to make use of managing outbound connections using banIP, for the purpose of mitigating having clients contacting botnet control servers and other rogue hosts.

My observation is, that the table 'banIP' is referenced 4 times:
```
forwarding_lan_rule
forwarding_wan_rule
input_lan_rule
input_wan_rule
```

The table 'banIP' consists only rules, wich are containing the wan device in all cases, either as 'in' or 'out' interface, depending on whatever you have selected in the Web-UI (src, dst, src+dst). Because of that, you'll never have a match using the 'input_lan_rule' reference. For obvious reasons the wan interface doesn't play a role on lan interface input rules. Thus the reference to the table 'input_lan_rule' is obsolete and should be removed in the default setup.

While the use of 'forwarding_lan_rule' fits my mentioned goal, the router itself is unprotected in that regard. The control of the output chain on the wan interface is completely missing here.

It would make more sense to me, having the table 'banIP' referenced like this:
```
forwarding_lan_rule
forwarding_wan_rule
input_wan_rule
output_wan_rule
```

Like this we would be able to protect the router itself, inbound AND outbound wise, against the wan side. I know that covers a rare scenario, but it simply makes sense to me.
Having 'input_lan_rule' running through all the 'banIP' lines is just waste of CPU and time.

dibdot, 2021-07-24:
Currently I do not plan to change the defaults. In banIP 0.7.x (comes with OpenWrt 21.x). You can freely assign (multiple) chains and interfaces - to fulfill your needs. banIP 0.3.x in OpenWrt 19.07 is an early bird and not really supported anymore.

closed 2021-07-24 by dibdot

manuel-kraus-19, 2021-07-24:
Workaround, just for reference:

Re-assign 'input_lan_rule' with 'output_wan_rule'.