openwrt/packages #16218: all: packages repo ifname 2 months warning

wulfy23, 2021-07-24:
Maintainer: @BKPepe 
Environment: all

Description:

its been almost 2 months since ifname changes to master source code...

```
fgrep -r ifname . | grep network | cut -d'/' -f1,2,3 | sort | uniq | wc -l
24
```
roughly 24 packages are still referencing ifname ... has anything been done to address this?

BKPepe, 2021-07-24:
I am not the maintainer.

champtar, 2021-07-24:
Are 24 packages broken, or just using deprecated notation ? I encourage you to better describe the problem and maybe how to fix it instead of just ranting. You could even send PR ;)

Andy2244, 2021-07-24:
yeah whats this about?
Just ping the maintainers for the packages that need changes.

obsy, 2021-07-25:
```
grep -r type . | grep network | cut -d'/' -f1,2,3 | sort | uniq | wc -l
11

```

champtar, 2021-07-25:
Still doesn't answer the question, are the packages broken or just using deprecated notation ?

obsy, 2021-07-25:
Broken. For ex. see https://github.com/openwrt/packages/issues/16007

champtar, 2021-07-25:
I just tested zabbix wifi discovery and it's working fine (it's using ubus)
The issue is all the users of `uci get network.$net.ifname` or `uci_get_state`

The correct way to get the ifname seems to be `network_get_device ifname lan`

Here a list (with some false positive)
```
$ grep -rs uci feeds | grep ifname
feeds/routing/ahcpd/files/ahcpd.init:	local ifname=$(uci_get_state network "$1" ifname "$1")
feeds/routing/oonf-init-scripts/files/oonf_init.sh:    interface="$( uci -q -c /etc/config get "${DAEMON}.@[${i}].ifname" )" || {
feeds/routing/cjdns/files/cjdns.defaults:      ifname=`uci get network.lan.ifname`
feeds/routing/babeld/files/babeld.init:	local ifname=$(uci_get_state network "$_name" ifname "$_name")
feeds/packages/ipv6/tayga/files/tayga.sh:	uci_set_state network "$cfg" ifname $link
feeds/packages/net/tinc/files/tinc.init:				local ifname=$(uci -P /var/state get network.$v.ifname 2>&-)
feeds/packages/net/isc-dhcp/files/dhclient6.init:	/usr/sbin/dhclient -q -nw -6 -cf $config_file -lf $lease_file -pf $pid_file -sf $script_file `/sbin/uci get network.wan.ifname`
feeds/packages/net/isc-dhcp/files/dhclient.init:	/usr/sbin/dhclient -q -nw -cf $config_file -lf $lease_file -pf $pid_file -sf $script_file `/sbin/uci get network.wan.ifname`
feeds/packages/net/shorewall6-lite/files/hotplug_iface:DEVICE=${DEVICE:-$(/sbin/uci -p /var/state get network."$INTERFACE".ifname)}
feeds/packages/net/shorewall/files/hotplug_iface:DEVICE=${DEVICE:-$(/sbin/uci -p /var/state get network."$INTERFACE".ifname)}
feeds/packages/net/vpn-policy-routing/files/vpn-policy-routing.init:is_ovpn() { local dev; dev=$(uci -q get network."$1".ifname); [ "${dev:0:3}" = "tun" ] || [ "${dev:0:3}" = "tap" ] || [ -f "/sys/devices/virtual/net/${dev}/tun_flags" ]; }
feeds/packages/net/vpn-policy-routing/files/vpn-policy-routing.init:		if [ "$1" = "$(uci -q get "network.${n}.ifname" || echo "$n")" ] || \
feeds/packages/net/vpn-policy-routing/files/vpn-policy-routing.init:		dev="$(uci -q get network."${wanIface4}".ifname)"
feeds/packages/net/vpn-policy-routing/files/vpn-policy-routing.init:		dev6="$(uci -q get network."${wanIface6}".ifname)"
feeds/packages/net/nft-qos/files/lib/priority.sh:				ifname="$(uci_get network.lan.ifname)"
feeds/packages/net/nft-qos/files/lib/priority.sh:			ifname="$(uci_get network.$priority_netdev.ifname)"
feeds/packages/net/vnstat/files/vnstat.defaults:wan="$(uci -P/var/state get network.wan.ifname)"
feeds/packages/net/wg-installer/wg-server/hotplug.d/99-mesh-babeld:	uci set babeld.@interface[-1].ifname="${INTERFACE}"
feeds/packages/net/wg-installer/wg-server/hotplug.d/99-mesh-babeld:		if [ "$(uci get babeld.@interface[$i].ifname)" == "${INTERFACE}" ]; then
feeds/packages/net/shorewall6/files/hotplug_iface:DEVICE=${DEVICE:-$(/sbin/uci -p /var/state get network."$INTERFACE".ifname)}
feeds/packages/net/shorewall-lite/files/hotplug_iface:DEVICE=${DEVICE:-$(/sbin/uci -p /var/state get network."$INTERFACE".ifname)}
feeds/packages/net/vpnbypass/files/vpnbypass.init:is_ovpn() { local dev; dev=$(uci -q get network."$1".ifname); if [ "${dev:0:3}" = "tun" ] || [ "${dev:0:3}" = "tap" ] || [ -f "/sys/devices/virtual/net/${dev}/tun_flags" ]; then return 0; else return 1; fi; }
feeds/luci/protocols/luci-proto-relay/htdocs/luci-static/resources/protocol/relay.js:		    deviceNames = L.toArray(uci.get('network', this.sid, 'ifname')),
feeds/luci/docs/jsapi/network.js.html:				var ifnames = L.toArray(uciInterfaces[i].ifname);
feeds/luci/docs/jsapi/network.js.html:		var ifnames = L.toArray(uci.get('network', this.sid, 'ifname')),
feeds/luci/docs/jsapi/network.js.html:			var ifnames = L.toArray(uci.get('network', this.sid, 'ifname'));
feeds/luci/docs/jsapi/network.js.html:		var ifnames = L.toArray(uci.get('network', this.sid, 'ifname'));
feeds/luci/docs/jsapi/network.js.html:		var ifnames = L.toArray(uci.get('network', this.sid, 'ifname'));
feeds/luci/applications/luci-app-mwan3/luasrc/model/cbi/mwan/interface.lua:				local device = uci:get("network", iface, "ifname")
feeds/luci/applications/luci-app-splash/root/etc/init.d/luci_splash:	parentiface="$(uci -q get network.${net}.ifname)"
feeds/luci/applications/luci-app-splash/root/etc/init.d/luci_splash:	ipt_log "prerouting_${zone}_rule" -i "${ifname%:*}" -s "$NETWORK/$PREFIX" -j luci_splash_prerouting -t nat
feeds/luci/applications/luci-app-splash/root/etc/init.d/luci_splash:	ipt_log "forwarding_${zone}_rule"    -i "${ifname%:*}" -s "$NETWORK/$PREFIX" -j luci_splash_forwarding -t filter
feeds/luci/applications/luci-app-splash/root/etc/init.d/luci_splash:		ipt6_log "forwarding_${zone}_rule"    -i "${ifname%:*}" -s "$ip6addr" -j luci_splash_forwarding -t filter
feeds/luci/applications/luci-app-dockerman/luasrc/model/docker.lua:	uci:set("network", dev_name, "ifname", device)
feeds/luci/applications/luci-app-dockerman/luasrc/model/docker.lua:	uci:set("network", if_name, "ifname", dev_name)
feeds/luci/applications/luci-app-cshark/luasrc/controller/cshark.lua:		luci.util.shellquote(ifname),
feeds/luci/applications/luci-app-cshark/luasrc/view/cshark.htm:			csxhr.get('<%=luci.dispatcher.build_url("admin", "network")%>/cshark_iface_dump_start/' + ifname + '/' + tx_val + '/' + type_val + '/'+ filter_val, null,
feeds/luci/modules/luci-base/htdocs/luci-static/resources/network.js:				var ifnames = L.toArray(uciInterfaces[i].ifname);
feeds/luci/modules/luci-base/htdocs/luci-static/resources/network.js:		var ifnames = L.toArray(uci.get('network', this.sid, 'device')),
[... cut ...]
feeds/luci/modules/luci-mod-admin-mini/luasrc/model/cbi/mini/wifi.lua:			m.uci:set("network", "wan", "_ifname", oldif)
feeds/luci/modules/luci-mod-admin-mini/luasrc/model/cbi/mini/wifi.lua:		m.uci:set("network", "wan", "ifname", " ")
feeds/luci/modules/luci-mod-admin-mini/luasrc/model/cbi/mini/wifi.lua:		if m.uci:get("network", "wan", "_ifname") then
feeds/luci/modules/luci-mod-admin-mini/luasrc/model/cbi/mini/wifi.lua:			m.uci:set("network", "wan", "ifname", m.uci:get("network", "wan", "_ifname"))
```

champtar, 2021-08-12:
Actually all the users of `uci_get_state` and `uci -p /var/state get` are fine, so not that much breakage in the end I think

wulfy23, 2021-08-13:
yggdrasil as of a few weeks ago inserts uci ifname config forcing a second/subsequent luci config migration

professor-jonny, 2021-09-15:
So as I under stand it a stantax change has been made in etc/confic/network under config interface Ifname has been changed to device among a list of other small changes?

could someone point me to a comit for another app that has been updated to work with the new config so I may look at how it has been fixed so I may try to fix nft-qos as I'm not very skilled in coding?

The maintainer of this app has not made any comits since 2019 and does not seem to be contactable and her web site is offline since ages also. I dont mind trying to look at it as it is an app I would really like to use.

champtar, 2021-09-15:
> So as I under stand it a stantax change has been made in etc/confic/network under config interface Ifname has been changed to device among a list of other small changes?
>
> could someone point me to a [...]

I might have fixed it here but need someone to run test it https://github.com/openwrt/packages/pull/16350

stangri, 2021-09-16:
> The correct way to get the ifname seems to be `network_get_device ifname lan`

Is this backwards compatible? How far back?

champtar, 2025-06-28:
This was either fixed or not impacting anyone, let's close this