openwrt/packages #16219: Softethervpn doesn't compile  libmayaqua.so: UNDEFINED REFERENCE to ENGINE_by_id

desigabri, 2021-07-25:
**Can't compile SoftEtherVPN-5.02.5180  server**

Maintainer: @fededim
Environment: ipq806x.  Archer C2600  21.02-rc3

Description:

_[ 98%] Linking C executable ../../vpnserver
/home/gabriele/openwrt/21/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ../../libmayaqua.so: undefined reference to `ENGINE_load_private_key'
/home/gabriele/openwrt/21/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ../../libmayaqua.so: undefined reference to `ENGINE_by_id'
/home/gabriele/openwrt/21/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: ../../libmayaqua.so: undefined reference to `ENGINE_init'
collect2: error: ld returned 1 exit status
make[7]: *** [src/vpnserver/CMakeFiles/vpnserver.dir/build.make:106: vpnserver] Error 1
make[7]: Leaving directory '/home/gabriele/openwrt/21/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/SoftEtherVPN-5.02.5180'
make[6]: *** [CMakeFiles/Makefile2:460: src/vpnserver/CMakeFiles/vpnserver.dir/all] Error 2
make[6]: Leaving directory '/home/gabriele/openwrt/21/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/SoftEtherVPN-5.02.5180'
make[5]: *** [CMakeFiles/Makefile2:467: src/vpnserver/CMakeFiles/vpnserver.dir/rule] Error 2
make[5]: Leaving directory '/home/gabriele/openwrt/21/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/SoftEtherVPN-5.02.5180'
make[4]: *** [Makefile:270: vpnserver] Error 2
make[4]: Leaving directory '/home/gabriele/openwrt/21/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/SoftEtherVPN-5.02.5180'
make[3]: *** [Makefile:176: /home/gabriele/openwrt/21/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/SoftEtherVPN-5.02.5180/.built] Error 2
make[3]: Leaving directory '/home/gabriele/openwrt/21/feeds/packages/net/softethervpn5'
time: package/feeds/packages/softethervpn5/compile#0.61#0.43#0.96
    ERROR: package/feeds/packages/softethervpn5 failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/softethervpn5/compile] Error 1
make[2]: Leaving directory '/home/gabriele/openwrt/21'
make[1]: *** [package/Makefile:108: /home/gabriele/openwrt/21/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gabriele/openwrt/21'
make: *** [/home/gabriele/openwrt/21/include/toplevel.mk:230: world] Error 2_

desigabri, 2021-07-25:
forcing previous SoftEtherVPN version, compilation goes right, so it's something due to the new package sources.

thanks in advance

neheb, 2021-08-03:
What host platform are you compiling on?

desigabri, 2021-08-04:
Debian 10 on Virtual BOX

neheb, 2021-08-08:
It looks like you have OpenSSL engines disabled?

neheb, 2021-08-08:
https://github.com/neheb/packages/commit/a9a2c4555f5133d3125db1d59380a22375c1a9c0 has a fix.

neheb, 2021-08-08:
https://github.com/SoftEtherVPN/SoftEtherVPN/commit/ff3910eb869f567ac9c57fea0135619a0affb27a indicates that this patch is correct.