openwrt/packages #16222: HTTPS-DNS-PROXY doesn't let router works in WAN is disconnected

desigabri, 2021-07-25:
Maintainer: @stangri

Environment: Qualcomm Atheros IPQ806X, Archer C2600 v1, "21.02, Master

Description:

 I found the trouble was https-dns-proxy and not of others packages. I had to uninstall it to have my routers working well back and after that remove the package from firmware images.

 The problem show itself when there isn't connection at the WAN port.  The router become inacessible from LAN.
 I tried just disabling the service but anyway it is enabled at boot by default. So the only way I had to solve this trouble was to forget it. Because I couldn't understand if it is due to any kind of incompatibility issue with nginx service I opened tickets for all two in the past. Now I can say that the matter is https-dns-proxy.

stangri, 2021-07-26:
Thanks for your feedback. Could you please elaborate on:

> The router becomes inacessible from LAN.

Please describe in detail what's happening. Does it not provide IP via DHCP, does it not reply to SSH/WebUI request, how are you accessing it (name or IP address)?

> The problem shows itself when there isn't connection at the WAN port.

Have you installed `https-dns-proxy` on the router which permanently does not have WAN access? Is your WAN access flakey or do you manually bring WAN interface down every now and then?

> I tried just disabling the service but anyway it is enabled at boot by default.

I've just tested on OpenWrt 21.02.0-rc3 and `service <servicename> disable` or `/etc/init.d/servicename disable` doesn't actually disable any services. However `chmod -x /etc/init.d/servicename` seems to work. This issue doesn't seem to be `https-dns-proxy` exclusive.

desigabri, 2021-07-26:
Thanks for your feedback. Could you please elaborate on:

    _The router becomes inacessible from LAN._

**Please describe in detail what's happening. Does it not provide IP via DHCP, does it not reply to SSH/WebUI request, how are you accessing it (name or IP address)?**

1) BOOT with WAN cable disconnected:
It provides IP
Login page OK but after login it becomes stuck (web access)
It is possible acces by SSH

2)Connect Cable to WAN:
 it starts working after less then 1 minute

3) disconnect cable from WAN
it loose web access after some time 1 minute or little more.

4) BOOT with WAN cable connected 
all goes right

    _The problem shows itself when there isn't connection at the WAN port._

**Have you installed https-dns-proxy on the router which permanently does not have WAN access? Is your WAN access flakey or do you manually bring WAN interface down every now and then?**

I use routers connected to WAN but I test firmwares with router alone and disconnected (I don't want lose internet access if firmware has problems). I use also router portable to connect to AP sharing internet on LAN and wifi and I can't access it and configure it without WAN. it's frustrating

    _I tried just disabling the service but anyway it is enabled at boot by default._

**I've just tested on OpenWrt 21.02.0-rc3 and service <servicename> disable or /etc/init.d/servicename disable doesn't actually disable any services. However chmod -x /etc/init.d/servicename seems to work. This issue doesn't seem to be https-dns-proxy exclusive.**

I'm able to disable the service if the router is already connected to WAN but I used to try the firmwares onto alone routers before installing the firmware onto working ones.

stangri, 2021-07-27:
I can't reproduce, I pulled the WAN cable out of my router, waited a minute or so and could still access WebUI just fine. I'll try to find the time to experiment with the WAN cable being disconnected for more than a few minutes sometime later this week.

stangri, 2021-07-29:
@desigabri Do you know how to use SDK to build an individual package? If I were to build one for you, do you know how to download and install it on your router?

desigabri, 2021-07-29:
I think I'll be able to install It if you give me the package. I never used
SDK but I compiled packages with the firmware. Will It Be different if
compiled trought SDK?

closed 2021-08-02 by stangri

desigabri, 2021-08-04:
hi stangri, I just compiled firmware and  now I can connect to the modem also without WAN connected, thanks

PR #16252 (2021-08-02): https-dns-proxy: update to 2021-07-29-1
Maintainer: me
Compile tested: x86_64, Sophos SG-105, OpenWrt 21.02.0-rc3
Run tested: x86_64, Sophos SG-105, OpenWrt 21.02.0-rc3, start/stop/name resolution/log output

Description:
* update binary to the latest commit (2021-07-29) to fix #16222 and #16239
* add hotplug.d/iface file and update Makefile to install it
* use Cloudflare's and Google's bootstrap DNS if bootstrap DNS is missing
* minor improvements in append_bool function
* add append_counter function for verbosity setting
* add append_bootstrap function (and supporting functions) to parse/san [... cut]
PR #16253 (2021-08-02): [21.02] https-dns-proxy: update to 2021-07-29-1
Maintainer: me
Compile tested: x86_64, Sophos SG-105, OpenWrt 21.02.0-rc3
Run tested: x86_64, Sophos SG-105, OpenWrt 21.02.0-rc3, start/stop/name resolution/log output

Description:
* update binary to the latest commit (2021-07-29) to fix #16222 and #16239
* add hotplug.d/iface file and update Makefile to install it
* use Cloudflare's and Google's bootstrap DNS if bootstrap DNS is missing
* minor improvements in append_bool function
* add append_counter function for verbosity setting
* add append_bootstrap function (and supporting functions) to p [... cut]
commit 062e8f4f (2021-08-02): https-dns-proxy: update to 2021-07-29-01

* update binary to the latest commit (2021-07-29) to fix #16222 and #16239
* add hotplug.d/iface file and update Makefile to install it
* use Cloudflare's and Google's bootstrap DNS if bootstrap DNS is missing
* minor improvements in append_bool function
* add append_counter function for verbosity setting
* add append_bootstrap function (and supporting functions) to parse/sanitize bootstrap setting
* move firewall array from 'main' instance to the first proxy instance
* delete useless 'main' instace
commit 3cb1894b (2021-08-02): https-dns-proxy: update to 2021-07-29-01

* update binary to the latest commit (2021-07-29) to fix #16222 and #16239
* add hotplug.d/iface file and update Makefile to install it
* use Cloudflare's and Google's bootstrap DNS if bootstrap DNS is missing
* minor improvements in append_bool function
* add append_counter function for verbosity setting
* add append_bootstrap function (and supporting functions) to parse/sanitize bootstrap setting
* move firewall array from 'main' instance to the first proxy instance
* delete useless 'main' instace