openwrt/packages #16223: 需要您的帮忙。

dengshen3328, 2021-07-26:
Hi, nice to meet you. I am from China and I need your help. I don’t understand the computer, I want to install for the router: openconnect client
config interface'MYVPN'
Router model: mt7621-xiaomi_redmi-router-ac2100
Can you help me?
Installed firmware: openwrt-ramips-mt7621-xiaomi_redmi-router-ac2100-squashfs-sysupgrade

Hope we can become friends and come to our China one day to play.

openwrtdiy, 2021-07-26:
首先进入路由器后台，然后点击 系统 > 软件包
接着就进入软件包的页面后，点击“更新列表”按钮。
最后选择可用列表中插件的“luci-proto-openconnect”进行安装。如果需要LuCI界面的话，还可以再安装“luci-app-ocserv”插件。

dengshen3328, 2021-07-26:
按照您的方法，菜单页面没有出现“MyVPN”出来。可以帮我远程看看吗？

发自我的iPhone

 
首先进入路由器后台，然后点击 系统 &gt; 软件包

dengshen3328, 2021-07-29:
> 安装好插件后，进入路由器后台界面上选择 网络 &gt; 接口
>  选择左下角的 “添加新接口”按钮，之后会弹出窗口设置名称“MYVPN”，协议选择“OpenConnect”，再点创建接口。
>  接着在窗口中设置相应的VPN参数及指定防火墙关联接口，“保存”按钮。
>  最后一步选择 “保存并应用”按钮

不知道为什么，每次保存后关电源后都没有保存到，也连不上，您那边可以操作试试看吗？我可以把路由器寄给您看看吗？

发自我的iPhone

openwrtdiy, 2021-07-31:
你的固件是官方OpenWrt固件？还是Lienol或lean的固件？如果固件是你自己编译的话，可以在编译的时候选择好luci-proto-openconnect插件。另外，新建VPN接口时要**保存两次**设置。VPN连不上，你要检查VPN客户端参数和VPN服务器是否可以访问。如果VPN服务器是国外容易被封禁。

dengshen3328, 2021-07-31:
您可以帮忙编译一次吗

发自我的iPhone

dengshen3328, 2021-08-04:
是OpengWRT固件，现在是可以保存，但就是上不了网。不知道怎么设置防火墙之类的，怎么办呢？

发自我的iPhone

BKPepe, 2021-10-31:
Hi, we can provide some help only in English and we are missing details like:
- which version of OpenWrt are you using
- which router do you have
- how are you trying to install the package
etc.

Try to ask [community](https://forum.openwrt.org/) how to proceed further. In the meantime, I'm closing this.

closed 2021-10-31 by BKPepe

dengshen3328, 2021-10-31:
which version of OpenWrt are you using→→OpenWrt R21.7.15

which router do you have→→→→→→→→Redmi AC2100

how are you trying to install the package
etc.→→→→→→→→Where can I download and install it on the openwrt official website.

Do you want to close this community forum?

If it is closed, how can I contact you later?

Do you have a Facebook account? This is convenient for contact.

Will you be willing to help others in the future?