openwrt/packages #16232: mosquitto-nossl lacks mosquitto_passwd utility

condemil, 2021-07-27:
Maintainer: @karlp
Environment: arm_cortex-a9_neon 21.02.0-rc3

Description:

As per #1909 the `mosquitto_passwd` utility was added for `mosquitto-ssl` package, but not to the `mosquitto-nossl`.

karlp, 2021-07-27:
Sure, sounds reasonable enough, in the world where no ssl is reasonable at least .... :)

karlp, 2021-08-03:
Actually, this simply isn't possible.  The mosquitto_passwd utility _uses_ openssl libraries as part of it's functionality.  If you want/need some subset of this, please file a ticket upstream with them about removing this limitation.  See https://github.com/eclipse/mosquitto/blob/master/apps/mosquitto_passwd/mosquitto_passwd.c#L23-L24  I'm closing this for now.

closed 2021-08-03 by karlp