openwrt/packages #16233: keepalived: unable to build with json support

gweiss76, 2021-07-27:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @feckert 
Environment:  OpenWrt openwrt-21.02 branch 55d9c020a1d7bcf1f7dceebfc9417de1eb448d41
- packages feed: aacfbe1534f505c9afeba9148f34affd8a1e9829
- architecture is: ramips/mt7621 - xiaomi router 4a

Description:
When building keepalive with the below config it fails. Building without JSON support is fine.

build log arround failure:
```
mv -f .deps/vrrp_ip_rule_route_parser.Tpo .deps/vrrp_ip_rule_route_parser.Po
mipsel-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../../lib  -I ../../keepalived/include -I ../../lib -D_GNU_SOURCE -I/build/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/build/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/build/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include  -I/build/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/build/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/build/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include  -D_GNU_SOURCE -g -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/build/build_dir/target-mipsel_24kc_musl/keepalived-2.2.1=keepalived-2.2.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.132 -Wall -Wextra -Wunused -Wstrict-prototypes -Walloca -Walloc-zero -Warray-bounds=2 -Wattribute-alias -Wbad-function-cast -Wcast-align -Wcast-qual -Wchkp -Wdate-time -Wdisabled-optimization -Wdouble-promotion -Wduplicated-branches -Wduplicated-cond -Wfloat-conversion -Wfloat-equal -Wformat-overflow -Wformat-security -Wformat-signedness -Wformat-truncation -Wframe-larger-than=5120 -Wimplicit-fallthrough=3 -Winit-self -Winline -Wjump-misses-init -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wmissing-prototypes -Wnested-externs -Wnormalized -Wnull-dereference -Wold-style-definition -Woverlength-strings -Wpointer-arith -Wredundant-decls -Wshadow -Wshift-overflow=2 -Wstack-protector -Wstrict-overflow=4 -Wstrict-prototypes -Wstringop-overflow=2 -Wsuggest-attribute=cold -Wsuggest-attribute=const -Wsuggest-attribute=format -Wsuggest-attribute=malloc -Wsuggest-attribute=noreturn -Wsuggest-attribute=pure -Wsync-nand -Wtrampolines -Wundef -Wuninitialized -Wunknown-pragmas -Wunsuffixed-float-constants -Wunused-const-variable=2 -Wunused-macros -Wvariadic-macros -Wwrite-strings -fPIE -Wformat -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -O2  -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/build/build_dir/target-mipsel_24kc_musl/keepalived-2.2.1=keepalived-2.2.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.132  -MT vrrp_json.o -MD -MP -MF .deps/vrrp_json.Tpo -c -o vrrp_json.o vrrp_json.c
<command-line>: warning: "_FORTIFY_SOURCE" redefined
<command-line>: note: this is the location of the previous definition
vrrp_json.c: In function 'vrrp_json_auth_dump':
vrrp_json.c:137:11: error: 'vrrp_t' {aka 'struct _vrrp_t'} has no member named 'auth_type'
  if (!vrrp->auth_type)
           ^~
vrrp_json.c:140:18: error: 'vrrp_t' {aka 'struct _vrrp_t'} has no member named 'auth_data'
  memcpy(buf, vrrp->auth_data, sizeof(vrrp->auth_data));
                  ^~
vrrp_json.c:140:42: error: 'vrrp_t' {aka 'struct _vrrp_t'} has no member named 'auth_data'
  memcpy(buf, vrrp->auth_data, sizeof(vrrp->auth_data));
                                          ^~
vrrp_json.c:141:17: error: 'vrrp_t' {aka 'struct _vrrp_t'} has no member named 'auth_data'
  buf[sizeof(vrrp->auth_data)] = 0;
                 ^~
At top level:
vrrp_json.c:133:1: warning: 'vrrp_json_auth_dump' defined but not used [-Wunused-function]
 vrrp_json_auth_dump(json_writer_t *wr, const char *prop, vrrp_t *vrrp)
 ^~~~~~~~~~~~~~~~~~~
make[6]: *** [Makefile:423: vrrp_json.o] Error 1
make[6]: Leaving directory '/build/build_dir/target-mipsel_24kc_musl/keepalived-2.2.1/keepalived/vrrp'
make[5]: *** [Makefile:572: all-recursive] Error 1
make[5]: Leaving directory '/build/build_dir/target-mipsel_24kc_musl/keepalived-2.2.1/keepalived'
make[4]: *** [Makefile:438: all-recursive] Error 1
make[4]: Leaving directory '/build/build_dir/target-mipsel_24kc_musl/keepalived-2.2.1'
make[3]: *** [Makefile:276: /build/build_dir/target-mipsel_24kc_musl/keepalived-2.2.1/.built] Error 2
make[3]: Leaving directory '/build/feeds/packages/net/keepalived'
time: package/feeds/packages/keepalived/compile#10.74#2.35#11.97
    ERROR: package/feeds/packages/keepalived failed to build.
```

openwrt config:
```
$ grep KEEPALIVE .config
CONFIG_KEEPALIVED_BFD=y
# CONFIG_KEEPALIVED_SHA1 is not set
# CONFIG_KEEPALIVED_LVS is not set
CONFIG_KEEPALIVED_VRRP=y
# CONFIG_KEEPALIVED_IPTABLES is not set
# CONFIG_KEEPALIVED_SNMP_VRRP is not set
# CONFIG_KEEPALIVED_SNMP_RFC2 is not set
# CONFIG_KEEPALIVED_SNMP_RFC3 is not set
# CONFIG_KEEPALIVED_DBUS is not set
CONFIG_KEEPALIVED_JSON=y
# CONFIG_KEEPALIVED_VRRP_AUTH is not set
# CONFIG_KEEPALIVED_CHECKSUM_COMPAT is not set
CONFIG_KEEPALIVED_ROUTES=y
CONFIG_KEEPALIVED_LINKBEAT=y
```

gweiss76, 2021-07-27:
building keepalived with CONFIG_KEEPALIVED_VRRP_AUTH=y the build finishes, but has missing dependencies later...

```install -m0755 ./files/etc/uci-defaults/keepalived /build/build_dir/target-mipsel_24kc_musl/keepalived-2.2.1/ipkg-mipsel_24kc/keepalived/etc/uci-defaults/keepalived
install -d -m0755 /build/build_dir/target-mipsel_24kc_musl/keepalived-2.2.1/ipkg-mipsel_24kc/keepalived/etc/init.d
install -m0755 ./files/keepalived.init /build/build_dir/target-mipsel_24kc_musl/keepalived-2.2.1/ipkg-mipsel_24kc/keepalived/etc/init.d/keepalived
install -d -m0755 /build/build_dir/target-mipsel_24kc_musl/keepalived-2.2.1/ipkg-mipsel_24kc/keepalived/etc/config
install -m0600 ./files/keepalived.config /build/build_dir/target-mipsel_24kc_musl/keepalived-2.2.1/ipkg-mipsel_24kc/keepalived/etc/config/keepalived
install -d -m0755 /build/build_dir/target-mipsel_24kc_musl/keepalived-2.2.1/ipkg-mipsel_24kc/keepalived/etc
install -m0600 ./files/keepalived.user /build/build_dir/target-mipsel_24kc_musl/keepalived-2.2.1/ipkg-mipsel_24kc/keepalived/etc/keepalived.user
install -d -m0755 /build/build_dir/target-mipsel_24kc_musl/keepalived-2.2.1/ipkg-mipsel_24kc/keepalived/etc/hotplug.d/keepalived
install -m0644 ./files/hotplug-user /build/build_dir/target-mipsel_24kc_musl/keepalived-2.2.1/ipkg-mipsel_24kc/keepalived/etc/hotplug.d/keepalived/01-user
find /build/build_dir/target-mipsel_24kc_musl/keepalived-2.2.1/ipkg-mipsel_24kc/keepalived -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package keepalived is missing dependencies for the following libraries:
libcrypto.so.1.1
libssl.so.1.1
make[3]: *** [Makefile:296: /build/bin/packages/mipsel_24kc/packages/keepalived_2.2.1-1_mipsel_24kc.ipk] Error 1
make[3]: Leaving directory '/build/feeds/packages/net/keepalived'
time: package/feeds/packages/keepalived/compile#13.45#2.86#15.05
    ERROR: package/feeds/packages/keepalived failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/keepalived/compile] Error 1
make[2]: Leaving directory '/build'
make[1]: *** [package/Makefile:108: /build/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/build'
make: *** [/build/include/toplevel.mk:230: world] Error 2
```

gweiss76, 2021-07-27:
compiling keepalived with a plain .config with just adding JSON support did work. so its my config.
I will check do some more tests and report back.

feckert, 2021-07-27:
We have to update the DEPENDS https://github.com/openwrt/packages/blob/master/net/keepalived/Makefile#L61 with `ssl` and `crypto` if we build with `CONFIG_KEEPALIVED_VRRP_AUTH`

gweiss76, 2021-07-27:
Hi Florian. This is not a full analysis, but i'm almost sure that the issue lies somewhere here:

I require freeradius with SSL support (only openssl). This takes me down the road of a openssl/wolfssl issue.
Trouble starts for me when i add luci-ssl-openssl which has a conflict with libustream-wolfssl20201210.
I have removed libustream-wolfssl20201210 and libwolfssl (moving all packages to openssl).
I cannot pinpoint the config setting right now. I will go on and rebuild/compile until i have it.

gweiss76, 2021-08-29:
Hi Florian. During debugging of this issue I recreated my openwrt configuration from scratch and the issue went away. I do not have a clue what went wrong here. Please feel free to close this issue (not sure if you want to fix the DEPENDS as stated above) and sorry for the noise.

feckert, 2021-08-30:
@gweiss76 Thanks for your feedback and for testing the keepalived with json. Once you have narrowed down the problem, can you please create a pull request? So we could fix the runtime dependency for CONFIG_KEEPALIVED_VRRP_AUTH?