openwrt/packages #16239: https-dns-proxy: uses 100% of cpu when no internet

Azq2, 2021-07-28:
[image]

Strace:
```
...a lot of logs like this...
epoll_pwait(3, [], 64, 0, NULL, 16)     = 0
open("/etc/hosts", O_RDONLY|O_LARGEFILE) = 8
read(8, "127.0.0.1 localhost\n\n::1     loc"..., 1024) = 110
read(8, "", 1024)                       = 0
close(8)                                = 0
socket(AF_INET, SOCK_DGRAM, IPPROTO_IP) = 8
fcntl64(8, F_GETFL)                     = 0x2 (flags O_RDWR)
fcntl64(8, F_SETFL, O_RDWR|O_NONBLOCK|O_LARGEFILE) = 0
fcntl64(8, F_SETFD, FD_CLOEXEC)         = 0
connect(8, {sa_family=AF_INET, sin_port=htons(53), sin_addr=inet_addr("1.1.1.1")}, 16) = -1 ENETUNREACH (Network unreachable)
close(8)                                = 0
socket(AF_INET, SOCK_DGRAM, IPPROTO_IP) = 8
fcntl64(8, F_GETFL)                     = 0x2 (flags O_RDWR)
fcntl64(8, F_SETFL, O_RDWR|O_NONBLOCK|O_LARGEFILE) = 0
fcntl64(^C8, F_SETFD, FD_CLOEXECstrace: Process 2269 detached
...
```

I think, its need some timeout between tries. 

P.S. may be on some low-end routers it also causes issues like this: #16222 (not sure)

stangri, 2021-07-29:
@Azq2 can you build the package using this [Makefile](https://github.com/stangri/source.openwrt.melmac.net/blob/master/https-dns-proxy/Makefile) and retest?

If not, let's just wait until the fixes are merged in the principal https_dns_proxy repo so they can be integrated in the OpenWrt repo package.

Azq2, 2021-07-29:
> @Azq2 can you build the package using this [Makefile](https://github.com/stangri/source.openwrt.melmac.net/blob/master/https-dns-proxy/Makefile) and retest?
>
> If not, let's just wait until the fixes [...]

Works fine with latest package, 0% cpu usage when no internet

stangri, 2021-07-29:
> P.S. I tested with https://github.com/aarond10/https_dns_proxy master

Would be great if you could also build/test with the (not yet merged) @baranyaib90's latest commit if you have the time. That's why I posted the Makefile pointing to it. Thanks!

stangri, 2021-08-02:
Should be fixed by #16252

closed 2021-08-02 by stangri

desigabri, 2021-08-04:
hi stangri, I just compiled firmware and  now I can connect to the modem also without WAN connected, thanks