openwrt/packages #16240: mwan3: Port forwarding fails on wana but works on wanb when mwan3 is started

fizzyduck, 2021-07-28:
Maintainer: @feckert @aaronjg
Environment: ARMv7, Linksys WRT1900ACv2, OpenWrt 19.07.7 r11306-c4a6851c72 / LuCI openwrt-19.07 branch git-21.044.30835-34e0d65
mwan3 version: 2.8.16-1

Description:

If I set up a port forward with Network->Firewall->Port Forwards from the wan zone to an internal IP, it works fine when mwan3 is not running (and connecting to wana public IP) but after running `mwan3 start` the port forward is only established when attempting to connect to wanb's public IP and not wana's public IP.

If I monitor with conntrack -E I can see with mwan3 not running I get a TCP packet ESTABLISHED, but after starting mwan3, I get a CLOSE in place of ESTABLISHED (I can see a mark of 256 also).

I note when running tcpdump that when mwan3 is running, the packets are appearing on the wrong interface which may be the issue here,

I am happy to provide logs as needed to help resolve this issue.

feckert, 2021-08-17:
As described in the [wiki](https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#interface_configuration) have you added the uci option `flush_conntrack` to the interface section?

fizzyduck, 2021-08-19:
Hi @feckert. Yes, I did try this but it has no effect at all and the connection is still not established.

fizzyduck, 2021-09-08:
@feckert I am happy to provide logs to help debug this issue if you can provide me with some assistance on which logs I should generate :)

feckert, 2021-09-08:
Sorry but this is not the normal use case for the mwan3.
I must say I don't have such a setup running.
It is just an incoming connection and not an outgoing one.
Why are you using the mwan3 in your setup at all?

To narrow down the error, `the /etc/config/mwan3` would not be bad and the output of `mwan3 internal ipv4` or `mwan3 internal ipv6`.

closed 2025-11-14 by GeorgeSapkin